bambulab/BambuStudio #3855: Auto-Orient Not Auto Orienting

jebsimmons, 2024-04-09:
### Bambu Studio Version

1.9.0.69

### Where is the application from?

Bambu Lab github releases

### OS version

14.4.1 (23E224)

### Printer

X1C

### How to reproduce

1. Import .stl
2. Press Auto-Orient Button
3. See that it doesn't Auto Orient

### Actual results

No action is taken on the object.

### Expected results

For the object to flip to the best orientation. 

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2024-04-12:
The object's orientation is already the best according to the algorithm. Of course, algorithms can make mistakes. You may share the model to help us improve.

jebsimmons, 2024-04-12:
> The object's orientation is already the best according to the algorithm. Of course, algorithms can make mistakes. You may share the model to help us improve.

I can't share the model- it's IP. This in the image is a lid- the top is flat an in the current production release it lays the flat side on the bottom. In this beta release it does not. @capsel22 can confirm this is happening with multiple models. The algorithm is not choosing the optimal print layout.

jebsimmons, 2024-05-05:
Seems to be fixed in that latest Public Beta 1.9.1.58

closed 2024-05-05 by jebsimmons