bambulab/BambuStudio #3856: Object Management

jebsimmons, 2024-04-09:
Positioning multiple objects with different colors requires drag and drop in the object view option. 

Currently when I load a .STL file that has inlays built in, I load the file and then press "Split to Objects"

I then go through each object and assign a color to the object. When finished, I select all the objects and press "Assemble"

I then have to manually drag each object to the right level in order to be able to have it print correctly. So in the assembly if I have one large item that is black and it's the last object in the list, I have to manually click that object and drag it up to the top. This is okay when I have 4 or 5 objects, but with hundreds it can be time consuming placing it at the right "level".

Having a system in place where you could easily move multiple objects at a time to the right location, or "Assign a level" to each object. Could be a table that lists all the objects and the color assigned and then you could manually put a level in.

Also, when all the objects are one color, the outline that gets applied to the object when you click on it is hard to identify the object in the view. the white box outline woks great on larger objects, but difficult to identify smaller objects.

philchfree, 2024-04-16:
You can use ctrl key and select multiple objects. Would this resolve your issue?

jebsimmons, 2024-04-16:
> You can use ctrl key and select multiple objects. Would this resolve your issue?

@philchfree In the object view when you select on one and press cmd/ctrl you can select multiple objects to change color- but when you try to drag multiple objects to the bottom- it deselects one of the objects. 

Therefore, if I have a black object and it is hidden by the following objects and I need it to drag it down to the bottom. That's fine for one, but with 20+ objects it is difficult to make sure it's at the right layer.