bambulab/BambuStudio #3857: Random Plate Changes

uniforx, 2024-04-09:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.4

### Additional system information

M2 Max 64GB

### Printer

Bambu Lab P1S

### How to reproduce

1. Create a new project
2. Choose a special plate e.g. Engineered Plate
3. Save the project
4. Doing many things with other projects .. after some days .. sometimes the Bambu Studio collabsed
5. Open the project and some times it´s now a Textured PEI Plate

### Actual results

The plate changes randomly.

### Expected results

The plate stay like defined.

### Project file & Debug log uploads

The projects are confidential.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-04-10:
[image]
You can hold down the Ctrl+P shortcut key and then cancel the hot bed following function.

uniforx, 2024-04-10:
Okay ... just for my understanding and maybe others too ... why would that make sense? I usually always print a special project the same way on the same bed/plate

closed 2024-05-09 by Haidiye00