bambulab/BambuStudio #3859: Beta 1.9.0 beta 2 - Error "Voronoi cell doesn't contain a sourcepoint"

3dprintandpaint, 2024-04-10:
### Bambu Studio Version

1.9.0.69

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

AMD Ryzen 7 3700X 8-Core Processor, 3600 Mhz, 8 Core(s), 16 Logical Processor(s)
B450 TOMAHAWK MAX (MS-7C02)
AMD Radeon RX 5700 XT

### Printer

X1C

### How to reproduce

Trying to slice the following multicoloured model from Cults3d https://cults3d.com/en/3d-model/home/holy-quin

Error is reported within seconds of slicing and the file cannot be printed.  Does not happen in Orca.

### Actual results

[image]

### Expected results

Expect model to slice.

### Project file & Debug log uploads

[debug_Wed_Apr_10_06_35_08_2328.log.zip](https://github.com/bambulab/BambuStudio/files/14927982/debug_Wed_Apr_10_06_35_08_2328.log.zip)

Project is too big to upload, even zipped.

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-04-11:
@3dprintandpaint could you upload this 3mf to another place
or includes it in email sending to lane.wei@bambulab.com

thanks

3dprintandpaint, 2024-04-13:
Email sent 👍

zhimin-zeng-bambulab, 2024-04-24:
It has been fixed in v01.09.01.56, you can have a try.

closed 2024-04-24 by zhimin-zeng-bambulab