bambulab/BambuStudio #3860: 1.9.0 Public beta2 无法自动切换支撑耗材

SteamedCharger, 2024-04-10:
### Bambu Studio Version

1.9.0 Public beta2

### Where is the application from?

Bambu Lab github releases

### OS version

Windows11

### Additional system information

主板：技嘉B660i
CPU：i5 12600K
内存：光威DDR4 3200 16Gx2

### Printer

Bambu Lab X1C

### How to reproduce

1.在支撑耗材中将“支撑/筏层界面”设置为Sup.PLA，支撑类型普通(自动)
2.切片，打印

### Actual results

[image]
切片时一切正常，显示正确的耗材类型，但在发送打印任务窗口中没有支撑料选项，并且打印到支撑界面时没有换料动作

### Expected results

打印机在打印到支撑界面时应根据设置的“支撑/筏层界面”材料自动切换到支撑料打印

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14931212/log.zip)
[加湿器测试3 一体式浅水_1.zip](https://github.com/bambulab/BambuStudio/files/14931220/3._1.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2024-04-11:
@SteamedCharger 
thanks for reporting this issue
we have fixed this issue in beta3
please have a try at https://github.com/bambulab/BambuStudio/releases/tag/v01.09.00.70

XunZhangBambu, 2024-04-11:
@SteamedCharger 已在beta3修复

closed 2024-05-06 by XunZhangBambu