bambulab/BambuStudio #3863: Use last color for next layer

Dan203, 2024-04-10:
When doing a multicolor print the slicer should be smart enough to use the last color from the previous layer as the first color for the next layer to save a color change and purge. Right now it seems to always cycle through the colors in order regardless of which colors are used on a given layer. If you used the last color from the previous layer as the first color on the next layer it would save quite a few purges. Bonus points if it could be planned out so that the last color used was always a color needed for the next layer even if the next layer has fewer colors than the previous one.

SaltWei, 2024-04-11:
[image]
[image]
Seems to be ok as your suggestion with the latest version

Dan203, 2024-04-11:
Hmmm... I sat there and watched it print and saw it not do this. I was using the latest release version, not the beta though, so maybe this is a recent addition?

My model had a prime tower, could that affect it? I can't read G-Code so I'm not exactly sure what you're pointing out in your screenshots.

SaltWei, 2024-04-12:
> My model had a prime tower, could that affect it? I can't read G-Code so I'm not exactly sure what you're pointing out in your screenshots.

We also use the latest release version, and seems to be OK as well. Could you share your 3mf as zip file so we can help to check the color order?

zhimin-zeng-bambulab, 2024-04-16:
[image]
@Dan203  Have you specified the printing order of all layers here?
Or could you share your 3mf file so that we can better analyze the problem?

Dan203, 2024-04-17:
I just tested with the latest beta and it is working correctly, so either this has already been fixed or I was wrong and it wasn’t working the way I thought. 

Anyway sorry for the bad report, please close.

SaltWei, 2024-04-17:
Thank you very much!

closed 2024-04-17 by SaltWei