bambulab/BambuStudio #3865: Support Material Does Not Work

YnotThreeve, 2024-04-10:
### Bambu Studio Version

1.9.0.69

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS Sonoma 14.0

### Additional system information

Apple M2, 8GB RAM

### Printer

Bambu Lab X1C

### How to reproduce

Try to print either plate in this file, no support material will be available in the print dialog
Support material missing as filament type in print dialogue
[image]
[image]

### Actual results

When trying to print with support material, the support material does not appear in the material selection dropdowns during the print dialog

### Expected results

Support material should be available to select just like the other filaments. This is a new but major problem to 1.9

### Project file & Debug log uploads

[Bug Report.3mf.zip](https://github.com/bambulab/BambuStudio/files/14938779/Bug.Report.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2024-04-11:
@YnotThreeve 
thanks for reporting this issue
we have fixed it in beta3 https://github.com/bambulab/BambuStudio/releases/tag/v01.09.00.70
coule you have a try?

TinyMito, 2024-04-12:
I'm testing this now with 1.09.0070 :), and it seems to be working.

Previously with 1.09.0069
About 30 mins ago. The filament swap to Support filament was stuck, no activities. I switch it back to 1.8 and test it again, no issue.

skl111, 2024-04-14:
> @YnotThreeve thanks for reporting this issue we have fixed it in beta3 https://github.com/bambulab/BambuStudio/releases/tag/v01.09.00.70 coule you have a try?

FYI the A1 Mini does not show PA/PET support as an option to be selected.
[image]
[image]

closed 2024-06-26 by XunZhangBambu