bambulab/BambuStudio #3870: bugs when adding a custom filament

Prorok13, 2024-04-12:
### Bambu Studio Version

1.9.0.7

### Where is the application from?

Bambu Lab github releases

### OS version

Win 11

### Printer

X1 Carbon

### How to reproduce

1. add a plastic thread
[image]

2. At the bottom of the list, select add\remove filament

[image]

3. Click "Custom filaments" 4. Click "Create New"

[image]

[image]

5. choose a vendor

### Actual results

1. There is no previously added vendor
3. It is not possible to add a custom type of filament

### Expected results

1. The previously added manufacturer must be in the list.
3. It should be possible to add a custom filament type and/or select from a previously added

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14956602/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

pl4nkt0n, 2024-04-13:
Would call it "feature request" - anyway one up.

DanBao-Bambu, 2024-06-20:
The first requirement has been recorded, and if you want to add some custom types of Filament Types, you can contact us. After thorough testing, we will add Filament presets for this type.