bambulab/BambuStudio #3871: Top layer is printed in the air with Gyroid in fill pattern.

Panda631, 2024-04-12:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

X1C & 2xAMS

### How to reproduce

Create a model with Gyroid infill pattern or download this model from maker world (e.g. plate 5)
https://makerworld.com/en/models/105669#profileId-112806
After slicing, look at layer 207. The base for the square holes are printed into the air and results in problems.

[image]

### Actual results

The printed part is quite uneven and has mechanical problems

### Expected results

A reliable support structure should be built for that.

### Project file & Debug log uploads

3mf file is online available

### Checklist of files to include

- [ ] Log file
- [X] Project file

dragon-eyes-bbl, 2024-04-24:
hi @Panda631  could you try again in the latest 1.9.1 beta4 to see if it has been fixed? I cannot reproduce it with the latest release. 
From layer 207 to 210, i didnt find this layer. 
Let me know how is it. Thanks!

[image]

dragon-eyes-bbl, 2024-05-16:
Close as inactive. Please reopen if the problem presist in the new release.  Thanks

closed 2024-05-16 by dragon-eyes-bbl