bambulab/BambuStudio #3872: Don't get the requested top layers

LeftOu, 2024-04-12:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Intel R Xeon R
48 gig ram

### Printer

Bambu X1 Carbon

### How to reproduce

When I enter 3 for the "Top shell layers" I get 5 

### Actual results

I get 5 shell layers when I only want 3.  This happens all the time.

### Expected results

I should get 3 top shell layers.

### Project file & Debug log uploads

[Case.zip](https://github.com/bambulab/BambuStudio/files/14962077/Case.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

XunZhangBambu, 2024-04-16:
@LeftOu 
Hi, this is because you set the top_shell_thickness to 1mm. Given that the layer height is 0.2,  to meet the requirement of top_shell_thickness,there should be at lease 5 solid layers. If you only want 3 shell layers, set the top_shell_layers to 3 and top_shell_thickness to 3*layer_height.

SaltWei, 2024-05-20:
Not bug, and match current code logic

closed 2024-05-20 by SaltWei