bambulab/BambuStudio #3873: Full layer prints in a different sequence than before, causing ugly lines on top layer because it shows through

stineek, 2024-04-12:
### Bambu Studio Version

1.9.0.70

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

Bambu Lab X1C

### How to reproduce

[image]
Image on the left shows how it printed with the previous version of Bambu Studio (1.8), the left one shows the diagonal lines showing through (because silk PLA is sensitive like that). The layer before the top layer gets printed in sections and actuallu has overlapping lines too on the same layer, which causes bumps.

### Actual results

Look at the item on the right side. 
[image]
Here is how the layer below looks during printing. I stopped this one before it got to finish, which is why there are bigger gaps. The arrows show overlapping lines on the same layer.

### Expected results

The left item is printed the way it is intended, with a smooth top surface, and no overlapping lines below.

### Project file & Debug log uploads

[logfiles.zip](https://github.com/bambulab/BambuStudio/files/14962454/logfiles.zip)
[Error_tray.zip](https://github.com/bambulab/BambuStudio/files/14962458/Error_tray.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2024-04-13:
@stineek Thank you very much. We can reproduce this issue and will fix it ASAP

XunZhangBambu, 2024-04-18:
@stineek  We are handling the issue. To continue with your current printing, you can turn off “Only one wall on top surfaces”.

closed 2024-06-26 by XunZhangBambu