bambulab/BambuStudio #3874: Issue while determining linux memory size

lucianoloder, 2024-04-12:
### Bambu Studio Version

remotes/origin/HEAD -> origin/master, commit 9591d26398bec2f2c52a7348a74acfd87e4753b4

### Where is the application from?

Built from source by yourself or others

### OS version

Linux Mint 21.3 Virginia - kernel 5.15.0-101-generic #111-Ubuntu SMP Tue Mar 5 20:16:58 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

### Printer

hypercube evolution

### How to reproduce

While running the BuildLinux.sh script i noticed this error:

./BuildLinux.sh: line 17: [: -le: expected unary operator

It appears that the $FREE_MEM_GB variable is returning empty in my system, due the "grep 'Mem:'" command. My "free" implementation print an extra dot, and so this command dont work
$ free --version
free de procps-ng 3.3.17
$ free -g
               total       usada       livre    compart.  buff/cache  disponível
Mem.:             14           6           0           0           7           7
Swap:             17           1          16

Removing the ":" from grep solves the issue for me

The patch in this bug report solved the problem for me.

Thanks in advance.

### Actual results

./BuildLinux.sh: line 17: [: -le: expected unary operator

### Expected results

no error message

### Project file & Debug log uploads

[BuildLinux.sh.patch.zip](https://github.com/bambulab/BambuStudio/files/14962898/BuildLinux.sh.patch.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-04-16:
could you submit the patch into github directly?
just as some pull requests in https://github.com/bambulab/BambuStudio/pulls

lanewei120, 2024-04-19:
submit here https://github.com/bambulab/BambuStudio/pull/3912

closed 2024-04-19 by lanewei120