bambulab/BambuStudio #3876: Changing Filament With Height Range Modifier Does Not Work On Assemblies

MadMax389, 2024-04-13:
### Bambu Studio Version

1.9.0.7

### Where is the application from?

Bambu Lab github releases

### OS version

Win 11

### Additional system information

10th Gen Core i5 32GB

### Printer

Slicer - All Printers

### How to reproduce

Sometimes I use a height range modifier to change colors at certain layers. This feature works fine on models that are not assemblies. (It worked in v1.7 but not 1.8). If two models are assembled together, all modifiers in the height range seem to work EXCEPT the color change. To reproduce this problem:
1) Add a cube and a cylinder primitive to the plate.
[image]
2) Confirm that the height range modifier works the cube. I set a color change and infill change from 5mm to 10mm
[image]
3) Now assemble the cube to the cylinder and confirm that the height range modifier applies to the entire assembly
[image]

### Actual results

Notice that the infill=0 modification applies to both object, but the color change does not
[image]

### Expected results

Expected that the color change would apply to both objects just like the infill modifier. 

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-05-16:
It has been fixed in v01.09.01.66, please have a try.

closed 2024-05-16 by zhimin-zeng-bambulab

MadMax389, 2024-05-16:
Confirmed fixed, thank you.