bambulab/BambuStudio #3878: Alert on insufficient filament

ewilts, 2024-04-13:
Assume X1C printer, AMS, Bambu filament, and Bambu Studio

When a model is sliced or sent to the printer, I want to get an alert/warning if there is insufficient filament remaining to complete the job. The printer/AMS knows approximately how much filament is remaining on a spool from the RFID tags, and Bambu Studio knows how much filament the job will consume from each slot.

If there is insufficient filament when sending to the printer, report, for example, "Insufficient filament in AMS slot A4.  Require 800g, remaining 500g".  If the configuration allows auto-continuation on another spool, add "Remaining 300g will be taken from slot B1".  If no auto-continuation is available, ask the user if the print should be submitted anyway or if the print request should be cancelled.  The user may have another roll on the shelf to be used but may want to cancel the print submission if not.

This will help prevent users unexpectedly running out of filament during a print.

napalmz, 2024-04-19:
When a filament terminates and the case of autorefill does not occur (an equal filament is missing or has been deactivated), the printer pauses and allows you to restart using another filament in the AMS. Right?

ewilts, 2024-04-19:
Yes, the printer pauses but that doesn't help the user if another roll of desired filament is not available.