bambulab/BambuStudio #3883: Intermittent Corrupted Timelapse Video Files

GloinTheDark, 2024-04-14:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1

### How to reproduce

Print Any Model with timelapse enabled.

### Actual results

Some Timelapse videos are not playable. Various video players flag the files as corrupted.

Example of corrupted file...
https://github.com/bambulab/BambuStudio/assets/10621012/c056c9f0-217f-4fb7-bcfc-e3234a251ed4

Example of good file, for comparison (same model)...
https://github.com/bambulab/BambuStudio/assets/10621012/5cb24821-226f-4c50-af89-97e266202105

### Expected results

Uncorrupted video files for every print.

### Project file & Debug log uploads

Project file is not relevant. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

GloinTheDark, 2024-04-14:
I have tried playing the corrupted file with windows video player, Bambu Studio integrated player, IPhone player, Chrome browser, etc. They all report corrupted file.

ClawzSkunk, 2024-05-06:
I get the same issue [image]

I have tried a new SD card and formatting both cards repeatedly.