bambulab/BambuStudio #3885: GIVE P1 USERS "PRINTER PARTS"

Kingsidorak, 2024-04-14:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows

### Additional system information

11

### Printer

3x P1P & 3x P1S

### How to reproduce

Own a P1 series printer, and use Bambu Studio

### Actual results

Not being able to set the printer accessories in the software

### Expected results

I have been getting horrible results printing in PETG for the last year. I just found out it's because I put a hardened steel nozzle on some of my printers, and the setting I need to change is buried in the menus on the printer itself, where as people with the X1 series printers have the option right in the software.  [image] 
[image]

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2024-04-17:
Hi @Kingsidorak 
It seems that the device has not yet established a connection, and functions related to the printer cannot be used until the connection is established.

Kingsidorak, 2024-04-17:
Unless I am misunderstanding what you mean, I have to say that is absolutely wrong. I have 6x P1 printers right now, 5 of them with AMS. 3 of them have Hardened Steel, the other 3 do not. On top of this I have reinstalled Windows from a clean install at least twice since owning P1 printers, and have installed several versions of Bambu Studio, just as well as reset some of my printers a few times in order to optimize my workflow + to make sure they work with Home Assistant (which stopped due to their IPs changing)

I am constantly in the device tab due to all the printers, and AMS', and I am often trying to explore the program for features. I have NEVER had this feature in the device tab on any version of Bambu Studio.

It is implemented in Orca Slicer no matter which printer I have, but not currently implemented in a way that is useful for both printers that have the upgrade and those that don't. I have already submitted an error report/feature request with them.

walterwongbbl, 2024-04-29:
Hi @Kingsidorak 
Is the firmware version up to date? I would like you to check the firmware version of the printer. The printer parts options will be displayed based on whether the printer supports the Studio-side nozzle type setting.

Kingsidorak, 2024-04-29:
The current public beta version of the software displays the Printers Parts option, but it did not in the full public releases before.

closed 2024-10-17 by walterwongbbl