bambulab/BambuStudio #3888: Studio does not display PA profile of second printer without APP restart

longtimeago, 2024-04-15:
### Bambu Studio Version

1.8.3.89

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04.4

### Additional system information

CPU: Intel® Core™ i9-9880H CPU @ 2.30GHz × 16
Graphics: NVIDIA Corporation TU117GLM [Quadro T2000 Mobile / Max-Q]
RAM: 64,0 GiB

### Printer

Bambulab X1C Carbon Combo x2

### How to reproduce

1. Have 2 printers with PA profiles
2. Open new project with Printer 1 (Bimba)
3. Goto Device tab and see PA profile on AMS spool display
4. Switch the view to Printer 2 (Simba)
5. PA profile is NOT displayed for Printer 2 (Simba)
6. Restart application
7. Open Device tab
8. PA profile is displayed for Printer 2 (Simba)
9. Switch to Printer 1 (Bimba)
10. PA profile is NOT displayed for Printer 1 (Bimba)

### Actual results

From step 3 (Profiles are displayed for Bimba)
[image]
From step 5 (Profiles are NOT displayed for Simba)
[image]
From step 8 (Profiles are displayed for Simba)
[image]
From step 10 (Profiles are NOT displayed for Bimba)
[image]

### Expected results

Profiles are displayed when switching between printers without a need of restarting the application

Reproduces with just plain empty project

### Project file & Debug log uploads

Log Files: [logs.zip](https://github.com/bambulab/BambuStudio/files/14983975/logs.zip)
Project: [new_empty_project.zip](https://github.com/bambulab/BambuStudio/files/14983992/new_empty_project.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-04-23:
@longtimeago  Thanks for reporting this issue, we will investigate the cause of this issue.

longtimeago, 2024-08-06:
@BambulabRobot still valid