bambulab/BambuStudio #3889: PETG Support interface layer not working

seanmcguire413, 2024-04-16:
### Bambu Studio Version

1.9.0.69 BETA

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

X1C + 2 AMS

### How to reproduce

https://makerworld.com/en/models/45111?from=search#profileId-109309
for the dome plate 2
support/raft base PLA
set support/raft interface to PETG
interface spacing set to 0
support/object xy distance 0
slice 

### Actual results

the slice preview shows there are two colors and has 21 filament changes
but when I got to print there is no support interface and only one ams slot is offered as an option on my printer

### Expected results

it should give me an option to assign both filament slots for my AMS and then print with two filaments for the support interface.  This was working in the previous build.

### Project file & Debug log uploads

[Pokeball+v2 support interface issue.zip](https://github.com/bambulab/BambuStudio/files/14985695/Pokeball%2Bv2.support.interface.issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

TinyMito, 2024-04-16:
Can you download the latest Beta 3? It has been fixed.

seanmcguire413, 2024-04-16:
Confirmed fixed.  Thank you for the help!!

closed 2024-04-16 by seanmcguire413