bambulab/BambuStudio #3890: Crashing after coloring object (1.9.0 Public Beta 3)

victornpb, 2024-04-16:
### Bambu Studio Version

v01.09.00.70-20240410162052

### Where is the application from?

Bambu Lab github releases

### OS version

macOS Sonoma 14.0 (23A344)

### Additional system information

Macbook M1 Pro 16Gb

### Printer

X1C

### How to reproduce

1. Add any model to build plate, select it
2. Press the color tool button
3. Pain it 
4. Try returning using Esc, or clicking Return button on top left.

### Actual results

App crashes

### Expected results

Not crash and return to normal prepare view.
As a workaround if I go to preview tab then back to prepare it doesn't crash.

### Project file & Debug log uploads

[BambuStudio-2024-04-16-012948.ips.zip](https://github.com/bambulab/BambuStudio/files/14986918/BambuStudio-2024-04-16-012948.ips.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Haidiye00, 2024-04-16:
sorry,this bug  just has fixed.At present, you can use ESC key to exit the coloring tool, and we will release the beta 4 version as soon as possible

Haidiye00, 2024-04-24:
Hello, this issue has been fixed. Please use the latest version. Thank you~

Haidiye00, 2024-04-24:
Hello，obj import color function has been released, supporting vertex color and mtl color import. PNG color is currently not supported, please try the new version. thanks

closed 2024-05-06 by Haidiye00