bambulab/BambuStudio #3898: Redo should have ⇧⌘Z shortcut like every other Mac app ever created

coreyward, 2024-04-17:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

macOS

### Printer

X1C

### How to reproduce

1. Perform some operations
2. Hit ⌘Z to undo
3. Try hitting ⇧⌘Z out of muscle memory
4. See app continue to undo rather than redo
5. Use Edit > Redo via mouse repeatedly, only to learn that the undo/redo cache is extremely limited
6. Realize you just lost 20 minutes of fiddly work

### Actual results

⇧⌘Z performs an undo rather than a redo

### Expected results

⇧⌘Z should perform a redo, or at the absolute barest minimum should do nothing. But it is a standard shortcut baked into every Mac app by default, so I don't understand why Bambu has seen fit to change it to ⌘Y of all things…the only thing I know if it being used for is open history in web browsers.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

coreyward, 2024-08-09:
@lanewei120 It seems like @LiZ-Li-BBL hasn't had an opportunity to contribute in the last month. Is there someone else that this should be assigned to?

xp-Yang, 2024-08-14:
In most Windows applications Ctrl+Y functions as Redo. Mac use ⌘+⇧+Z for Redo. We will support both Ctrl+Y and Ctrl+Shift+Z for Redo to resolve this conflict in the next release.

LiZ-Li-BBL, 2024-08-14:
In most Windows applications Ctrl+Y functions as Redo. Mac use ⌘+⇧+Z for Redo. We will support both Ctrl+Y and Ctrl+Shift+Z for Redo to resolve this conflict in the next release.

dotspencer, 2025-01-23:
The redo keystroke mapping has been very annoying to me as well. It really should be ⌘+⇧+Z on macOS

PR #7345 (2025-06-26): Add macOS ⌘⇧Z redo shortcut
## Summary
- add Shift+Command+Z accelerator for redo on macOS
- show the updated shortcut in the keyboard shortcuts dialog

Fixes #3898
closed 2025-06-26 by lanewei120