bambulab/BambuStudio #3899: Bambu Studio Crashing When Slicing

docshiv, 2024-04-17:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro Version 22H2

### Additional system information

Processor: Intel(R) Core(TM) i7-14700K 3.40 GHz
Installed RAM: 32.0 GB (31.8 GB usable)
System type: 64-bit operating system, x64-based processor
Video Card: NVIDIA GeForce RTX 4060 Ti

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Import model "main helmet (textured) wing separated Wolverine D3 helmet Do3D.stl" File is 
Click "Slice plate"

### Actual results

Bambu Studio crashes at 25%. It says it's generating walls. 
[image]

### Expected results

The model should be sliced and ready to print.

### Project file & Debug log uploads

[bambu-studio.exe.14520.dmp](https://github.com/bambulab/BambuStudio/files/15004694/bambu-studio.exe.14520.dmp)
[crash_Tue_Apr_16_21_14_41_0_done.log](https://github.com/bambulab/BambuStudio/files/15004696/crash_Tue_Apr_16_21_14_41_0_done.log)
[bambu-studio.exe.30756.dmp](https://github.com/bambulab/BambuStudio/files/15004697/bambu-studio.exe.30756.dmp)
[crash_Tue_Apr_16_21_12_23_0_done.log](https://github.com/bambulab/BambuStudio/files/15004698/crash_Tue_Apr_16_21_12_23_0_done.log)
[main helmet (plain) wing separated Wolverine D3 helmet Do3D.zip](https://github.com/bambulab/BambuStudio/files/15004689/main.helmet.plain.wing.separated.Wolverine.D3.helmet.Do3D.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2024-04-19:
[image]
The la'test version [1.9.0.70](https://github.com/bambulab/BambuStudio/releases/tag/v01.09.00.70) seems to be OK. Could you have a try?

QingZhangBambu, 2024-04-19:
[image] hi, it seams that it crashed on clipper lib. if you rotate the model, will the model be sliced successfully? if you update the old version or latest version works? and you may try this method to seem if it works. #3318

ozgur640, 2024-04-26:
Try setting process affinity to 4 cores in task manager for bambustudio.exe, this helped me.

QingZhangBambu, 2024-04-28:
@ozgur640, as this method works, you may want to do a CPU test and install a new CPU

ozgur640, 2024-04-28:
> @ozgur640, as this method works, you may need to do a CPU test and install a new CPU

Hi thanks for the reply, My processor is i9 13900k and quite new. Some older programs (rocksmith etc) cant handle too many cpu cores, thats why i play with affinity if it crashes. May be its related to prusaslicer (which bambustudio is based on), my operating system (Win10) or my gpu (gtx1070).
Best Regards..

closed 2024-05-21 by QingZhangBambu