bambulab/BambuStudio #3901: Check build plate type before heating bed.

MrDB42, 2024-04-17:
**Is your feature request related to a problem? Please describe.**
I know this a "personal order of operations" type of thing, but it would be REALLY nice if the printer would verify which build plate has been installed, prior to heating the bed and warning the user that the sliced model build plate doesn't match what is in place.

This would be nice when going back and forth from the PEI Cool Plate and anything thing else, since the Cool Plate only needs ~35C temperature. Sometimes I forget to change the appropriate build plate in the slicer but have already put a different build plate in the printer.

Example: Put in the Cool Plate, but the slicer is set for the PEI Smooth High Temp. The printer will heat the bed to the appropriate temperature for High Temp plate (>35C) then it checks which plate is installed (mismatch because the Cool Plate is installed) and it throws an error. I then go back to the project, change to the Cool Plate. But then I have to wait for the bed to cool down.

**Describe the solution you'd like**
Just have the printer verify the installed build plate matches the sliced g-code first.

MrDB42, 2024-05-16:
Figured out the solution:

Cut:
```
;===== check scanner clarity ===========================
G1 X128 Y128 F24000
G28 Z P0
M972 S5 P0
G1 X230 Y15 F24000
;===== check scanner clarity end =======================

```
And put it below:
`M204 S10000 ; init ACC set to 10m/s^2
`

MrDB42, 2024-05-19:
I've been able to simplify it a bit more:
```
M204 S10000 ; init ACC set to 10m/s^2

_;===== check scanner clarity ===========================
G28
M972 S5 P0
G1 X230 Y15 F24000
;===== check scanner clarity end =======================_

;===== heatbed preheat ====================
```

One benefit I have noticed by putting this process first is I have reduced the false positive errors of the printer not recognizing the type of build plate installed.

drpeppershaker, 2025-01-14:
+1 for this. 

The first thing that should happen is homing, then initial build plate check.
THEN preheating and wiping.

As it stands, it's entirely possible to damage the nozzle with a poorly aligned build plate as the wipe action occurs before the bed is properly homed. If homing occurs before everything we eliminate this issue. And checking the build plate type will also add additional time savings.

MrDB42, 2025-01-15:
Actually, playing around and watching and experimenting, I suggested the following order of operations:
1. Warm the nozzle to previous filament softening temperature.
2. Wipe the nozzle.
3. Home the print head.
4. Check the build plate (aka: check scanner clarity).
Then do all the "voodoo that you do": heat the build plate, re-home the print head, etc., etc.

I've been trying to plow through the "Machine start G-code" to try to achieve this. But without proper documentation of all the Bambu specific g-code it's rather daunting and frustrating.