bambulab/BambuStudio #3909: Plate Sliced Files not saving the plate

nykanent2017, 2024-04-18:
### Bambu Studio Version

1.8.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

Processor:13th Gen Intel(R) Core(TM) i9-13900H   2.60 GHz
Ram: 64GB
System: 64 bit
Intel m330 graphics

### Printer

Bambu Labs X1C

### How to reproduce

Save a plate sliced file with the textured plate.
Close out of the project and reopen the slicer.
Change the plate selection to the cool plate.
Reopen the original textured plate sliced file.
See that the plate the .3mf file opens with is the last selected plate.

### Actual results

.3mf files open with whatever plate was last selected and not what plate was saved at the time of .3mf file creation. ie in the above case it opens with the cool plate instead of the textured plate.

### Expected results

.3mf file should open and load the plate type to be that was selected at the time of the files creation.
see the two attached examples.

### Project file & Debug log uploads

[plate sliced file examples.zip](https://github.com/bambulab/BambuStudio/files/15029359/plate.sliced.file.examples.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-04-19:
[image]
You can press the Ctrl+P shortcut key and then turn off this option.

nykanent2017, 2024-04-19:
Thank you! That solved the issue!

closed 2024-05-15 by Haidiye00