bambulab/BambuStudio #3911: 编辑预设-钢材丝设置中参数文本框无法编辑

yueduz, 2024-04-19:
### Bambu Studio Version

1.9.0.70

### Where is the application from?

Built from source by yourself or others

### OS version

archlinux

### Printer

Bambu A1

### How to reproduce

打开耗材丝设置窗口,随机的出现无法编辑参数.出现问题时重启软件可恢复编辑.

### Actual results

随机的出现无法编辑参数

### Expected results

可编辑的文本框

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/15031992/log.zip)
[未命名.3mf.zip](https://github.com/bambulab/BambuStudio/files/15031994/3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

yueduz, 2024-06-09:
[录屏 2024-06-09 16-51-58.webm](https://github.com/bambulab/BambuStudio/assets/1448159/5c023036-3ec9-4053-aec6-29ac5e6f5b80)
切换选项卡后参数不可编辑

bambu123, 2024-06-14:
Fixed in version 1.9.2.

closed 2024-06-14 by bambu123