bambulab/BambuStudio #3913: AMS not detected

finn-devs, 2024-04-19:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Macos 14.4.1 (23E224)

### Additional system information

M1 mac mini

### Printer

Bambu x1 carbon

### How to reproduce

The AMS is not detected suddenly. After months of working suddenly the AMS is no longer detected. No amount of troubleshooting resolved the issue. 

### Actual results

No AMS.

### Expected results

AMS should work.

### Project file & Debug log uploads

[debug_Wed_Apr_17_16_24_51_1670.log.0.pdf](https://github.com/bambulab/BambuStudio/files/15034181/debug_Wed_Apr_17_16_24_51_1670.log.0.pdf)

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2024-04-29:
@finn-devs 
Hi, I checked your logs and found no data from AMS. Can you please submit a support request? It's possible that there might be some issues with your AMS itself. Thank you very much for your feedback.   You can contact our after-sales service

finn-devs, 2024-04-29:
Hi,

After a firmware update the issue resolved itself.

Thanks,
Fred

closed 2024-10-17 by walterwongbbl