bambulab/BambuStudio #3916: P1S prints Wrong first layer filament

Madmike-1337, 2024-04-19:
### Bambu Studio Version

1.9.0.70

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 23H2

### Printer

P1P

### How to reproduce

Load model
Assign PLA
Add Raft
Assign Support PLA for interface
Print on P1P

### Actual results

Prints first layer in Support PLA

### Expected results

To print first layer in PLA

### Project file & Debug log uploads

[Tip Wedge.zip](https://github.com/bambulab/BambuStudio/files/15046240/Tip.Wedge.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

XunZhangBambu, 2024-04-22:
Are you slicing first, then printing the gcode.3mf on p1p? If so,it's because in gcode.3mf we missed the support filament.We have identified and fixed this issue in next version. Thanks!

Madmike-1337, 2024-04-22:
Yes, that is correct.

Thanks for fixing the issue!

Madmike-1337, 2024-04-22:
> Are you slicing first, then printing the gcode.3mf on p1p? If so,it's because in gcode.3mf we missed the support filament.We have identified and fixed this issue in next version. Thanks!

This is on a P1S also.

closed 2024-04-29 by Madmike-1337