bambulab/BambuStudio #3917: 包含高度修改器的特定模型会导致软件在切片时崩溃

moe001, 2024-04-20:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

（在多台电脑上复现，推测无关。）
10940x/32g/3090

### Printer

P1SC

### How to reproduce

1. 菜单-打开项目-加载这个文件
2. 点击**切片单盘**或切换标签页到预览以使程序进行切片

### Actual results

程序会进入无响应状态(和/或)窗口直接消失，偶发的会弹出一个内容大意为“程序异常(非法访问)，建议保存项目并重启”的，有软件风格的提示框，复现十多次只见过一次，没有截到图。似乎问题跟模型上的修改器有关，但这是加工该模型必须的。

### Expected results

正常生成切片

### Project file & Debug log uploads

[bug.zip](https://github.com/bambulab/BambuStudio/files/15046802/bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2024-04-22:
@moe001 
帮忙传一下log文件，crash的话，在log目录下一般会有dmp文件生成
log放在 C:\Users\user.name\AppData\Roaming\BambuStudio\log

moe001, 2024-04-22:
[log.zip](https://github.com/bambulab/BambuStudio/files/15067857/log.zip)
这四个文件是新开一个程序实例并直接打开该文件，如上述步骤操作后直指程序崩溃，在你所述目录里新产生的所有文件。

zhimin-zeng-bambulab, 2024-05-16:
已经在 v01.09.01.66 版本中修复

closed 2024-05-16 by zhimin-zeng-bambulab