bambulab/BambuStudio #3922: Program crashes when object is placed on the build plate

nmcandrews, 2024-04-21:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: Ryzen 7 5800x @ 3.8 GHz
Memory: 64 GB 
GPU: Nvidia GeForce RTX 3080

### Printer

Bambu Lab P1S

### How to reproduce

Open Bambu Lab. 
If build plate does not have checkering on the plate, the program has high failure probability. 
Open file from project OR insert a new simple object to the build plate.
Select the item. 

### Actual results

Program shows Windows loading circle on the cursor for ~2 seconds, after object is selected.
Program crashes, with no error message displayed. 

Attempts were made to resolve the issue by re-installing the program and had no effect. 

### Expected results

Build plate should retain classic checkered pattern, and not appear as a flat gray color. 
Program should not crash. 

### Project file & Debug log uploads

[Apr 21 2024 Logs.zip](https://github.com/bambulab/BambuStudio/files/15052148/Apr.21.2024.Logs.zip)
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2024-04-22:
To analyze DMP files, we need you to provide the following DLLs on your system:

C:\Windows\System32\DriverStore\FileRepository\nv_dispi.inf_amd64_84b2c943d6816eb7\nvoglv64.dll
C:\Windows\System32\ntdll.dll
C:\Windows\System32\kernel32.dll
C:\Windows\System32\KERNELBASE.dll
C:\Windows\System32\user32.dll
C:\Windows\System32\msvcp140.dll
C:\Windows\System32\ucrtbase.dll
C:\Windows\System32\win32u.dll

nmcandrews, 2024-04-22:
[Troubleshooting Files.zip](https://github.com/bambulab/BambuStudio/files/15065434/Troubleshooting.Files.zip)

> To analyze DMP files, we need you to provide the following DLLs on your system:
>
> C:\Windows\System32\DriverStore\FileRepository\nv_dispi.inf_amd64_84b2c943d6816eb7\nvoglv64.dll [...]

bambu123, 2024-04-23:
>	nvoglv64.dll!00007ffd3d9e5420()	未知	非用户代码。无法查找或打开 PDB 文件。
 	nvoglv64.dll!00007ffd3da8f2e3()	未知	非用户代码。无法查找或打开 PDB 文件。
 	nvoglv64.dll!00007ffd3da80203()	未知	非用户代码。无法查找或打开 PDB 文件。
 	nvoglv64.dll!00007ffd3d63a469()	未知	非用户代码。无法查找或打开 PDB 文件。
 	nvoglv64.dll!00007ffd3d62628c()	未知	非用户代码。无法查找或打开 PDB 文件。
 	nvoglv64.dll!00007ffd3d62921f()	未知	非用户代码。无法查找或打开 PDB 文件。
 	nvoglv64.dll!00007ffd3d628ddb()	未知	非用户代码。无法查找或打开 PDB 文件。
 	nvoglv64.dll!00007ffd3d8a5798()	未知	非用户代码。无法查找或打开 PDB 文件。
 	kernel32.dll!BaseThreadInitThunk()	未知	非用户代码。已加载符号。
 	ntdll.dll!RtlUserThreadStart()	未知	非用户代码。已加载符号。

bambu123, 2024-04-23:
ntdll.dll!NtQueryInformationThread()	未知	非用户代码。已加载符号。
 	KERNELBASE.dll!00007ffdaff00253()	未知	非用户代码。无法查找或打开 PDB 文件。
 	nvoglv64.dll!00007ffd3d8a3912()	未知	非用户代码。无法查找或打开 PDB 文件。
 	nvoglv64.dll!00007ffd3d62be45()	未知	非用户代码。无法查找或打开 PDB 文件。
 	nvoglv64.dll!00007ffd3d7b935e()	未知	非用户代码。无法查找或打开 PDB 文件。
 	nvoglv64.dll!00007ffd3d4108c6()	未知	非用户代码。无法查找或打开 PDB 文件。
 	opengl32.dll!00007ffd3fbf93aa()	未知	非用户代码。无法查找或打开 PDB 文件。
 	BambuStudio.dll!Slic3r::GUI::ImGuiWrapper::render_draw_data(ImDrawData * draw_data) 行 2452	C++	已加载符号。
 	BambuStudio.dll!Slic3r::GUI::ImGuiWrapper::render() 行 509	C++	已加载符号。
 	BambuStudio.dll!Slic3r::GUI::GLCanvas3D::render(bool only_init) 行 1998	C++	已加载符号。
 	BambuStudio.dll!Slic3r::GUI::GLCanvas3D::on_mouse(wxMouseEvent & evt) 行 3999	C++	已加载符号。
 	BambuStudio.dll!wxAppConsoleBase::CallEventHandler(class wxEvtHandler *,class wxEventFunctor &,class wxEvent &)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxEvtHandler::ProcessEventIfMatchesId(struct wxEventTableEntryBase const &,class wxEvtHandler *,class wxEvent &)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxEvtHandler::SearchDynamicEventTable(class wxEvent &)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxEvtHandler::TryBeforeAndHere(class wxEvent &)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxEvtHandler::ProcessEvent(class wxEvent &)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxEvtHandler::SafelyProcessEvent(class wxEvent &)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxWindow::HandleMouseMove(int,int,unsigned int)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxWindow::MSWHandleMessage(__int64 *,unsigned int,unsigned __int64,__int64)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxWindow::MSWWindowProc(unsigned int,unsigned __int64,__int64)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxWndProc(struct HWND__ *,unsigned int,unsigned __int64,__int64)	C++	非用户代码。已加载符号。
 	user32.dll!00007ffdb1a4ef75()	未知	非用户代码。无法查找或打开 PDB 文件。
 	user32.dll!00007ffdb1a4e9ee()	未知	非用户代码。无法查找或打开 PDB 文件。
 	opengl32.dll!00007ffd3fc0f0f0()	未知	非用户代码。无法查找或打开 PDB 文件。
 	user32.dll!00007ffdb1a4ef75()	未知	非用户代码。无法查找或打开 PDB 文件。
 	user32.dll!00007ffdb1a4e69d()	未知	非用户代码。无法查找或打开 PDB 文件。
 	user32.dll!00007ffdb1a4bc66()	未知	非用户代码。无法查找或打开 PDB 文件。
 	BambuStudio.dll!wxWindow::MSWProcessMessage(struct tagMSG *)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxGUIEventLoop::PreProcessMessage(struct tagMSG *)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxGUIEventLoop::ProcessMessage(struct tagMSG *)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxGUIEventLoop::Dispatch(void)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxEventLoopManual::DoRun(void)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxEventLoopBase::Run(void)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxAppConsoleBase::MainLoop(void)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxEntryReal(int &,wchar_t * *)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!wxEntry(int &,wchar_t * *)	C++	非用户代码。已加载符号。
 	BambuStudio.dll!Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams & params) 行 64	C++	已加载符号。
 	BambuStudio.dll!Slic3r::CLI::run(int argc, char * * argv) 行 1023	C++	已加载符号。
 	BambuStudio.dll!bambustu_main(int argc, wchar_t * * argv) 行 5697	C++	已加载符号。
 	bambu-studio.exe!wWinMain(HINSTANCE__ * __formal, HINSTANCE__ * __formal, wchar_t * __formal, int __formal) 行 308	C++	已加载符号。

bambu123, 2024-04-23:
@nmcandrews Can you help try version 1.9?

nmcandrews, 2024-04-23:
I don't think I'd be of much help ... I'm basically brand new to this program and to 3D printing.. so I wouldn't know an issue if I saw it

ozgur640, 2024-04-26:
Try setting affinity to 4 cores in task manager for bambu studio, this helped me.

nmcandrews, 2024-04-29:
> setting affinity to 4 cores in task manager

It's already set for that... 
[image]

nmcandrews, 2024-04-29:
I just  added a new filament to my filaments list, and now the issue has returned.

nmcandrews, 2024-04-29:
[Troubleshooting Files - Set 2.zip](https://github.com/bambulab/BambuStudio/files/15155966/Troubleshooting.Files.-.Set.2.zip)

Haidiye00, 2024-04-30:
Hello, may I ask if your computer is a Windows or Mac computer

nmcandrews, 2024-04-30:
> Hello, may I ask if your computer is a Windows or Mac computer

Please view the original post at the top of this page. 
[image]

nmcandrews, 2024-05-03:
[image]
Deleted BOTH folders in %appdata%/Roaming (AKA - C:\Users\[username here]\AppData\Roaming) and STILL got the same error, after basically clearing all settings and preferences... Completely wiped all printers, filaments, settings, and profiles. No change to issue persistence.................

nmcandrews, 2024-05-05:
@bambu123

nmcandrews, 2024-05-08:
@bambu123 Have you looked into this more?

bambu123, 2024-05-08:
@nmcandrews Can you provide more recent DMP logs?
@Haidiye00  Please help analyze them.

nmcandrews, 2024-05-08:
> @nmcandrews Can you provide more recent DMP logs? @Haidiye00 Please help analyze them.

The dump logs listed under the post above are the most recent : [Troubleshooting Files - Set 2.zip](https://github.com/bambulab/BambuStudio/files/15155966/Troubleshooting.Files.-.Set.2.zip)
[04-29-2024 Crash Log Files.zip](https://github.com/bambulab/BambuStudio/files/15156128/04-29-2024.Crash.Log.Files.zip)

nmcandrews, 2024-05-11:
Opened the program randomly today to see if anything magically fixed itself, and it has... Suuuuper odd... I haven't changed a single thing since I submitted the last crash logs... I haven't reinstalled the program.. I haven't cleared anything, or modified a single thing. The only thing that I can think that changed was my Nvidia graphics driver version, which I updated yesterday... As of today, it just magically works again... The build plate has the checkered pattern again, and I can put stuff on the build plate without the program crashing... This bug is insane... 
[image]

nmcandrews, 2024-05-18:
Aaaand the demon is back... I was looking in my device settings to change a color on the AMS, and exited that tab to find the build plate didnt have checkering - which usually indicates the program failure / impending crash... 

I have a lot of stuff open on my PC at the moment (browser tabs and video editing software), and since I was just trying to configure a new spool of filament and do a calibration, and get that going while I edited, I did not take note of the build plate status prior to going into the device tab or configuration menus. 

In my experience - 
If the build plate doesnt have the grid / checkering, the program is going to crash. 
If the build plate does have the grid / checkering, the program will run without issue.
[image]

nmcandrews, 2024-05-18:
Bambu Studio Version
1.9.1.66

Where is the application from?
Bambu Lab Official website

OS version
Windows 11

Additional system information
CPU: Ryzen 7 5800x @ 3.8 GHz
Memory: 64 GB
GPU: Nvidia GeForce RTX 3080

Printer
Bambu Lab P1S

nmcandrews, 2024-05-18:
Something odd experienced -- 
Like I mentioned in the post before last, I was going to try a re-install of the graphics driver the next time (this time) that this particular issue occurred...   I decided to try something before I went full uninstall mode... 

I went to the toolbar, right clicked Nvidia's GeForce Experience program and EXITed the program. With the program still remaining closed, I tried to load a known-good project file. When the Studio program launched everything seemed normal and fine, but when the project file tried to load (i.e. - When the graphic was about to render on-screen inside the project plate area) the program completely crashes without loading the contents of the project file.

nmcandrews, 2024-05-19:
Tried a simple re-install of the Nvidia display drivers. Failed to correct the issue.

nmcandrews, 2024-05-19:
[image]
So weird ... 

..program even crashes just sitting here looking at the empty build plate.. doesn't even need to have anything on it...

nmcandrews, 2024-05-19:
...and still no help received since Day 1

nmcandrews, 2024-07-25:
Issue persists into version 1.9.3.5
Computer had to be hard-shutdown (hold power button until shutdown) earlier today, because things were running ungodly slow. I had been printing stuff just fine all day long, I do a hard-shutdown and bring it back online, now Bambu Studio AND OrcaSlicer fail to load the build environment without crashing. 
I am able to open every single tab EXCEPT FOR the prepare tab. As soon as the build plate and build environment start to load, the program freezes for 0.1 seconds and crashes.

nmcandrews, 2024-07-25:
[CrashLogs 07252024.zip](https://github.com/user-attachments/files/16384135/CrashLogs.07252024.zip)

nmcandrews, 2024-09-13:
Never ceases to amaze me just how many issues I can have with a program that works fine one minute, and is unresponsive and crashing the next minute. 
Today I was printing a file, and it printed fine. I left the application open, because I was going to look into some issues with the model, regarding percentage quantity of infill. I go outside, I get the mail, I open my mail. I tab back into the program and suddenly it's going into the classic Windows "This program is being unresponsive" translucent screen with the circling windows spiral cursor. Next thing I know it crashes. I open the program back up, from the desktop icon, and go to click on the "recent project" file for that thing I was just looking at, and it says that now it suddenly and magically can't find the file... Irritated, I go to File-Open and see the file sitting there in my downloads........ ..... .... .... I try to open it, yeah sure buddy... Opens the file, and the build plate is absent of the grid marks, like in the examples above, and the examples attached here..... ...you can see, not even an hour before, it still had them... now it doesnt.. and now the program crashes every single time I try to load anything...... just like before.... Program crashes within 3 seconds of loading ANY FILE. 
[image]
[image]
[image]
[image]
[image]

I'm getting quite irritated that this is a re-occurring issue that nobody seems to give a damn to look into or fix... I've been posting about this since the beginning of the year, and nobody has bothered to do anything with all my information I have supplied to them..................... It feels like the equivalent of placing an order in-person with a cashier and then they just walk away and never come back to you.............. Like, what is the point of me making this ticket if nobody is going to use the info to help me fix the problem, or even look at it??!

nmcandrews, 2024-09-13:
[BS Troubleshooting Files - 09-13-2024.zip](https://github.com/user-attachments/files/16998468/BS.Troubleshooting.Files.-.09-13-2024.zip)
[DxDiag 09-13-2024.txt](https://github.com/user-attachments/files/16998469/DxDiag.09-13-2024.txt)

nmcandrews, 2024-09-25:
Still crashes with the new install of Bambu Studio Version 1.9.7.52

I can sit on the Home, Project (Blank page - no model loaded), and Calibration screens just fine. 
I can also access the SD card and saved timelapse files, and download/delete them with no issues. 
It's only once I navigate to Prepare or Preview pages that it crashes - i.e. When it has to load something that accesses whatever on the back-end that loads/renders any type of graphics or 3D architecture. I don't even have to have a model loaded for it to crash. It will crash just off a blank new-file view of the build plate, so whatever calculates the size, dimensions, or textures for that 3D environment is likely the root of the crash... 
[crash_Tue_Sep_24_20_18_34_0_done.log](https://github.com/user-attachments/files/17122507/crash_Tue_Sep_24_20_18_34_0_done.log)

nmcandrews, 2024-09-25:
I can confirm that this is still an issue with the new Bambu Studio **version 1.9.7.52**

I can sit on any other screen just fine without it crashing. I can also access the SD card thru the Bambu Studio interface, and download/delete files on the SD card just fine. The crash seems to only happen when **on the Prepare or Preview tabs**. It doesn't even have to be switching to a plate with a model loaded; it can be a completely blank build plate, and it'll still crash to desktop. There is something wrong with the loading of the _3D environment_ when switching to the Prepare or Preview tabs. 

[crash_Tue_Sep_24_20_18_34_0_done.log](https://github.com/user-attachments/files/17122535/crash_Tue_Sep_24_20_18_34_0_done.log)

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-04-16 by Haidiye00