bambulab/BambuStudio #3927: Please add Notes Tab in BambuStudio Project

giufini, 2024-04-22:
Very often the name of a file sent to the printer from BambuStudio is not enough to remember several printing parameters to consider before starting a new print (build plate, brand and type of filament and more), so it would be greatly appreciated to have the opportunity to write down all these extra information in a specific "Notes" tab of each BambuStudio project.

It would obviously be useful to have the possibility of recalling these extra notes once a file to print has been selected on the printer screen.

SaltWei, 2024-05-15:
> If you can remember it is there, there is a Notes section for text at the end of Process settings under Others. OrcaSlicer implements this a little better, giving Notes their own tab, right next to Others where it is a little easier see and to remember.

[image]
Here? Bambu also has

giufini, 2024-05-15:
> "Here? Bambu also has. By the way, this is original setting from prusa,
> not from orca."
>
> I never used PrusaSlicer, and early versions of Studio and OrcaSlicer did
> not have Notes. Orca provided Notes shortly after Bambu took over the
> Projects tab for MakerWorld, and it was implemented some months later in
> Studio (maybe v1.7?).
>
> Unless you have a high resolution screen and work in a large window, it is
> easy to forget that Studio notes even exist. You have to open the Others
> tab and then scroll down to the very bottom to find them. Orca gives Notes
> their own tab along with Quality, Strength, Speed, etc., where they are
> easily accessed.

Hi LJKraus, you are right! I found the Notes field at the very bottom of
the "Others" tab.
Anyway, is it still possible to view these notes on the X1 printer screen
when selecting the file to print?

kfbest, 2025-09-15:
related to https://github.com/bambulab/BambuStudio/issues/2774