bambulab/BambuStudio #3930: Filament won't load from AMS, but loads fine by hand.

Ray-Peters-42, 2024-04-23:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

1. Goto Device. 2.In Control settings click "Load Filament".  

### Actual results

The printer heats the nozzle, then attempts to load.  It pushes the filament all the way to the extruder, then decides something isn't right and unloads.  repeats 4 or 5 times.   I can load single filament without AMS.

### Expected results

Filament should load and then print.

### Project file & Debug log uploads

[15 Tooth Sprocket.zip](https://github.com/bambulab/BambuStudio/files/15069968/15.Tooth.Sprocket.zip)
[debug_Fri_Apr_19_15_09_53_43108.log.zip](https://github.com/bambulab/BambuStudio/files/15069982/debug_Fri_Apr_19_15_09_53_43108.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Ray-Peters-42, 2024-05-04:
Are we going to discuss this, make any suggestions, or troubleshoot?

DataDivaNet, 2024-11-12:
When you filament by hand, do you feel any resistance from it right as it goes from the tube to the extruder?  And does that change a but by moving the bowden tube around?  It might be that you need a new filament sensor that's right above the extruder.  Luckily it's a cheap part and quick to replace.