bambulab/BambuStudio #3937: linux version bambu studio no flush volume

DatViciousLama, 2024-04-23:
### Bambu Studio Version

1.9.0.70

### Where is the application from?

Bambu Lab github releases

### OS version

linux mint 21.3

### Additional system information

System:
  Kernel: 5.15.0-102-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: Gigabyte product: Z390 UD v: N/A serial: <superuser required>
  Mobo: Gigabyte model: Z390 UD serial: <superuser required> UEFI: American Megatrends v: F11
    date: 12/21/2023
Battery:
  Device-1: hidpp_battery_0 model: Logitech Wireless Mouse MX Master 2S serial: <filter>
    charge: 55% (should be ignored) rechargeable: yes status: Discharging
CPU:
  Info: 8-core model: Intel Core i7-9700K bits: 64 type: MCP smt: <unsupported> arch: Coffee Lake
    rev: C cache: L1: 512 KiB L2: 2 MiB L3: 12 MiB
  Speed (MHz): avg: 852 high: 1217 min/max: 800/4900 cores: 1: 800 2: 800 3: 800 4: 800 5: 800
    6: 800 7: 800 8: 1217 bogomips: 57600
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: AMD Navi 22 [Radeon RX 6700/6700 XT / 6800M] vendor: Micro-Star MSI driver: amdgpu
    v: kernel pcie: speed: 16 GT/s lanes: 16 ports: active: HDMI-A-1 empty: DP-1,DP-2,DP-3
    bus-ID: 03:00.0 chip-ID: 1002:73df class-ID: 0300
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: amdgpu,ati
    unloaded: fbdev,modesetting,radeon,vesa gpu: amdgpu display-ID: :0 screens: 1
  Screen-1: 0 s-res: 6144x3456 s-dpi: 120 s-size: 1300x732mm (51.2x28.8") s-diag: 1492mm (58.7")
  Monitor-1: HDMI-A-0 mapped: HDMI-A-1 model: LG (GoldStar) HDR 4K serial: <filter>
    res: 6144x3456 hz: 60 dpi: 260 size: 600x340mm (23.6x13.4") diag: 690mm (27.2") modes:
    max: 3840x2160 min: 640x480
  OpenGL: renderer: AMD Radeon RX 6750 XT (navi22 LLVM 15.0.7 DRM 3.42 5.15.0-102-generic)
    v: 4.6 Mesa 23.2.1-1ubuntu3.1~22.04.2 direct render: Yes
Audio:
  Device-1: Intel Cannon Lake PCH cAVS vendor: Gigabyte driver: snd_hda_intel v: kernel
    bus-ID: 00:1f.3 chip-ID: 8086:a348 class-ID: 0403
  Device-2: AMD Navi 21 HDMI Audio [Radeon RX 6800/6800 XT / 6900 XT] driver: snd_hda_intel
    v: kernel pcie: speed: 16 GT/s lanes: 16 bus-ID: 03:00.1 chip-ID: 1002:ab28 class-ID: 0403
  Sound Server-1: ALSA v: k5.15.0-102-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: Gigabyte driver: r8169
    v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: 3000 bus-ID: 06:00.0 chip-ID: 10ec:8168
    class-ID: 0200
  IF: enp6s0 state: up speed: 1000 Mbps duplex: full mac: <filter>
Drives:
  Local Storage: total: 476.94 GiB used: 231.01 GiB (48.4%)
  ID-1: /dev/nvme0n1 vendor: Silicon Power model: SPCC M.2 PCIe SSD size: 476.94 GiB
    speed: 31.6 Gb/s lanes: 4 type: SSD serial: <filter> rev: T1103F0L temp: 31.9 C scheme: GPT
Partition:
  ID-1: / size: 467.89 GiB used: 231.01 GiB (49.4%) fs: ext4 dev: /dev/nvme0n1p2
  ID-2: /boot/efi size: 511 MiB used: 6.1 MiB (1.2%) fs: vfat dev: /dev/nvme0n1p1
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 0 KiB (0.0%) priority: -2 file: /swapfile
USB:
  Hub-1: 1-0:1 info: Hi-speed hub with single TT ports: 16 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
  Device-1: 1-2:2 info: Keychron K4 Pro type: Keyboard,HID driver: hid-generic,usbhid
    interfaces: 3 rev: 2.0 speed: 12 Mb/s power: 500mA chip-ID: 3434:0240 class-ID: 0300
  Device-2: 1-7:3 info: Logitech Unifying Receiver type: Keyboard,Mouse,HID
    driver: logitech-djreceiver,usbhid interfaces: 3 rev: 2.0 speed: 12 Mb/s power: 98mA
    chip-ID: 046d:c52b class-ID: 0300
  Hub-2: 2-0:1 info: Super-speed hub ports: 10 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003
    class-ID: 0900
Sensors:
  System Temperatures: cpu: 27.8 C pch: 35.0 C mobo: 16.8 C gpu: amdgpu temp: 42.0 C mem: 62.0 C
  Fan Speeds (RPM): N/A gpu: amdgpu fan: 0
Repos:
  Packages: 2370 apt: 2361 flatpak: 9
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com virginia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
Info:
  Processes: 317 Uptime: 3h 51m wakeups: 5 Memory: 31.26 GiB used: 5.67 GiB (18.1%) Init: systemd
  v: 249 runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12 Client: Cinnamon v: 6.0.4 inxi: 3.3.13

### Printer

x1c 

### How to reproduce

1. open program
2. open the flushing volumes window
3. change the multiplier from 0 to what works best for you (in my case .65)
4. press the recalculate button 
5. press ok as long as all flush volumes are black and not red
6. either set up a print, print a file, do anything with the program, ore just close/quit bambu studio
7. if you didn't close bambu studio in previous step do it now
8. once closed/quit program reopen program
9. open flushing volumes

### Actual results

flushing volumes all set/reset to zero.

### Expected results

flushing volumes should all remain set to previous settings.

### Project file & Debug log uploads

[Archive.zip](https://github.com/bambulab/BambuStudio/files/15077714/Archive.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MklBambu, 2024-05-06:
Hello, can you tell me your region and language of use?
Have you tried setting multiplier to 1 or higher?

DatViciousLama, 2024-05-08:
USA, English. Yes i did try to set the multiplier to 1 (i eventually settled on .65) but upon closing the software it reset to 0 when the software was relaunched. Thankfully the problem has since resolved itself. I am currently running 1.9.1.58 and all is back to normal.

MklBambu, 2024-05-17:
Hi, the new version has been released. You can try it out and verify if any issues still exist.
https://github.com/bambulab/BambuStudio/releases/