bambulab/BambuStudio #3940: Mark beta release as pre-release

strausmann, 2024-04-23:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Currently, both stable and beta releases of the BambuStudio are marked as normal versions in the GitHub releases. However, this means that it is not possible to distinguish between stable and beta releases via the GitHub API. Only the name of a release shows that it is a beta version.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I would like every BambuStudio release to be marked with the pre-release if it is a beta version. In the same way, I would like a new stable release to be marked as "latest release".

[image]

In addition, an unstable release, i.e. a beta version, should also be tagged with the version number and the suffix "-beta".

[image]

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

The solution has already been described above. This actually corresponds to a standard process for publishing software with stable and unstable versions on GitHub.

**Additional context**
Add any other context or screenshots about the feature request here.

lanewei120, 2024-04-26:
thanks for the suggestion

done for 1.9.1.58 beta 4
and will be done for all the future releases

strausmann, 2024-04-26:
Hello @lanewei120, I once drew a fork. If I want to create a release at the fork now, it looks as expected. 

[image]

 
Just adding a beta to the release tag won't make sure the GutHub API only delivers stable releases if you're looking for it. 

Please see again how you can mark the release as a prerelease. Thank you very much! 

That would improve the quality of my Chocolatey Package for the BambuStudio, in the update process. Currently, no distinction can be made between beta and stable. 

https://community.chocolatey.org/packages/bambustudio

 Thanks for the review and hopefully a better release experience.

strausmann, 2024-04-26:
With the Orca Slicer, for example, this is done correctly. Here you can distinguish between stable and pre-release

https://github.com/SoftFever/OrcaSlicer/releases

[image]

strausmann, 2024-05-21:
Is there any progress? I don't think it should be a big problem.

lanewei120, 2024-05-22:
> This ensures that the releases at Chocolatey can also be created as a pre-package.

do you mean change the previous version?

strausmann, 2024-05-22:
No, I mean whether we can do it in the future with pre-releases to mark them as such?

lanewei120, 2024-05-22:
yes, we can do it in the future versions

strausmann, 2024-06-09:
@lanewei120 

I have one more comment. Please do not simply change the assets in a GitLab release. If you have a problem with the v01.09.01.66 release on Windows, you cannot simply adapt v01.09.01.67 as assets for the v01.09.01.66 release. You don't do that. That would have been a completely new GitLab release.

strausmann, 2024-06-11:
@walterwongbbl 

With the new release, you have unfortunately forgotten to create a tag for this release. Could you please make sure that this tag matches the version in the release title?

[image]

Without the tag and thus the version number, no Chocolatey Package can be generated automatically for Bambu Studio.

lanewei120, 2024-06-11:
done
https://github.com/bambulab/BambuStudio/tree/v01.09.02.57

strausmann, 2024-06-11:
> done https://github.com/bambulab/BambuStudio/tree/v01.09.02.57

Thank you :-)