bambulab/BambuStudio #3941: 1.9.1 Public Beta 4 网络插件

wangda888, 2024-04-24:
### Bambu Studio Version

1.9.1 Public Beta 4

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.4.1 (23E224)

### Additional system information

M1 Pro 16G RAM 

### Printer

P1S

### How to reproduce

覆盖上个Beta3版本的安装包,然后打开App,提示网络插件无法下载,请检查VPN或防火墙

### Actual results

覆盖上个Beta3版本的安装包,然后打开App,提示网络插件无法下载,请检查VPN或防火墙

### Expected results

能够正常下载网络插件

### Project file & Debug log uploads

直接使用github下载的1.9.1 Public Beta 4 DMG安装包

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lichao0223, 2024-04-24:
遇到同样的问题

lanewei120, 2024-04-26:
我们更新了1.9.1 public beta4, 请更新一下image，
https://github.com/bambulab/BambuStudio/releases/tag/v01.09.01.58
并重新下载网络插件
给您带来不便，请见谅

closed 2024-04-26 by lanewei120