bambulab/BambuStudio #3947: Not uploading from Makerworld web site.

MaximusStalsky, 2024-04-24:
### Bambu Studio Version

1.9.0.70

### Where is the application from?

Bambu Lab github releases

### OS version

win 11

### Additional system information

5800x3d. 4080, 32gb ram, water cooled

### Printer

Bambu Lab x1 carbon ams

### How to reproduce

Just try to open any print from makerworld web site in bambu studio 

### Actual results

Nothing show up in the slicer.

### Expected results

The model should show up in the slicer

### Project file & Debug log uploads

[fail.zip](https://github.com/bambulab/BambuStudio/files/15097714/fail.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

walterwongbbl, 2024-04-29:
@MaximusStalsky 
[image]
I have loaded your 3mf file, which does not contain any model data. Where did I get this file from.

SaltWei, 2024-05-15:
[image]
Too small 3mf and can not open

Wait for response for a long time. Close.

Reopen if still has this problem

closed 2024-05-15 by SaltWei