bambulab/BambuStudio #3950: Missing network plugin for 1.9.0 beta 4

hadess, 2024-04-24:
### Bambu Studio Version

1.9.0 beta 4

### Where is the application from?

Built from source by yourself or others

### OS version

Linux x96-64 Flatpak

### Printer

N/A

### How to reproduce

The network plugin fails to load on startup with 1.9.0 beta 4.

Following https://wiki.bambulab.com/en/software/bambu-studio/failed-to-get-network-plugin, I try to access:
`https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.09.01.00`

### Actual results

```json
{"message":"success","code":null,"error":null,"software":{"type":null,"version":"01.08.04.51","description":"###https://wiki.bambulab.com/en/software/bambu-studio/release/release-note-1-8-4###","url":"https://public-cdn.bambulab.com/upgrade/studio/software/01.08.04.51/e5a4bb6a91/Bambu_Studio_win-v01.08.04.51.exe","force_update":false},"guide":null,"resources":[]}
```

### Expected results

Something with the URL to the cloud plugin

### Project file & Debug log uploads

You can reproduce by installing:
https://github.com/flathub/com.bambulab.BambuStudio/pull/20

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2024-04-25:
@hadess 
We are very sorry, but due to some reasons, we have temporarily taken down the 1.9 beta4. We will wait for it to be fixed and it will be re listed. Please continue to use Beta3 version

hadess, 2024-04-25:
Removing releases from https://github.com/bambulab/BambuStudio/releases doesn't send any notifications, and doesn't tell integrators like me why this particular version was removed.

Please make sure to think about and document how folks trying to distribute BambuStudio (like me for Flathub) get told about releases being "retracted", and make sure to add it to: https://github.com/bambulab/BambuStudio/issues/3395

lanewei120, 2024-04-26:
> Removing releases from https://github.com/bambulab/BambuStudio/releases doesn't send any notifications, and doesn't tell integrators like me why this particular version was removed.
>
> Please make [...]

We are very sorry
We will notify you as soon as possible if this situation occurs again

closed 2024-05-14 by lanewei120