bambulab/BambuStudio #3953: v01.09.01.58: No response after launch

kookxiang, 2024-04-25:
### Bambu Studio Version

1.9.1 Public Beta 4

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 Pro for Workstations 23H2

### Printer

Bambu Lab P1S

### How to reproduce

1. download and install from https://github.com/bambulab/BambuStudio/releases/tag/v01.09.01.58
2. open after install

### Actual results

[image]

### Expected results

expect it can create / open 3mf project

### Project file & Debug log uploads

[debug_Thu_Apr_25_22_28_06_15208.log](https://github.com/bambulab/BambuStudio/files/15111658/debug_Thu_Apr_25_22_28_06_15208.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

kookxiang, 2024-04-25:
BTW, the same version works on macOS Sonoma

bambu123, 2024-04-26:
Please use the method in the wiki to capture DMP when the application is stuck, and capture it several times.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

kookxiang, 2024-04-26:
After waiting for 2~3 minutes, the app finally started, and it happens each time.

Here's the DMP file during freeze and log file when it restored:
https://1drv.ms/u/s!AvoMTfx7AMtugb4ayQ8OyQhMZWggJg?e=gmbPoX
[debug_Fri_Apr_26_12_39_57_6772.log](https://github.com/bambulab/BambuStudio/files/15125485/debug_Fri_Apr_26_12_39_57_6772.log)