bambulab/BambuStudio #3956: Application crash running Release 1.9.1 Public Beta 4 (and 3)

tsmith35, 2024-04-26:
### Bambu Studio Version

1,9.1-beta4

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 22H2

### Additional system information

Lenovo P53 ThinkPad

### Printer

None (crashed before I could do anything)

### How to reproduce

Installed Studio 1.9.1-beta4 in normal C:\Program Files\Bambu Studio location.
Ran bambu-studio.exe from that location.
After 5-10 seconds of normal appearance while loading, the application abruptly crashes.

### Actual results

From Windows Event Viewer Application Log:

Faulting application name: bambu-studio.exe, version: 1.0.0.0, time stamp: 0x662a50c5
Faulting module name: ucrtbase.dll, version: 10.0.19041.3636, time stamp: 0x81cf5d89
Exception code: 0xc0000409
Fault offset: 0x000000000007286e
Faulting process id: 0x5fe4
Faulting application start time: 0x01da97b93c71bff2
Faulting application path: C:\Program Files\Bambu Studio\bambu-studio.exe
Faulting module path: C:\Windows\System32\ucrtbase.dll

### Expected results

Expecting program to run normally.

### Project file & Debug log uploads

No files to upload. Program crashes before I can do anything else.

### Checklist of files to include

- [X] Log file
- [ ] Project file

tsmith35, 2024-04-27:
Just wanted to add... I ran "dism /online /cleanup-image /restorehealth" and "sfc /scannow" to verify my OS integrity. All passed with no errors, but bambu-studio.exe still crashes just as before.

garry572, 2024-04-28:
Also having the same issue. Windows 10 Pro 22H2. All the versions I've tried do the same at random. unrelated operations
No other program crashes, OrcaSlicer doesn't crash, despite looking very similar
[image]

lanewei120, 2024-04-28:
@garry572 @tsmith35 

thanks for reporting these issues, could you help to upload the log file to us?
log files are in C:\Users\user.name\AppData\Roaming\BambuStudioBeta\log
and please includes the dmp files

Furthermore, Is the public beta3 crash on your machines?
https://github.com/bambulab/BambuStudio/releases/tag/v01.09.00.70

Haidiye00, 2024-04-28:
[image]
Can you try this method by deleting the corresponding item from the Windows registry.thanks

garry572, 2024-04-28:
Yes the beta 3 also does the same

garry572, 2024-04-28:
I've looked for the files above and only found a few of the registry keys. Tried to upload the log and dmp files but they were too big.

tsmith35, 2024-04-30:
> Can you try this method by deleting the corresponding item from the Windows registry.thanks

None of the registry items were found. Here are the log files and dump file. WinDbg says it's a buffer overflow.

[log.zip](https://github.com/bambulab/BambuStudio/files/15168661/log.zip)
[bambu-studio-dmp.zip](https://github.com/bambulab/BambuStudio/files/15168668/bambu-studio-dmp.zip)

tsmith35, 2024-05-01:
@lanewei120 were the log and dump files helpful?

tsmith35, 2024-05-03:
@bambu123 did you find any clues in the dump file?

tsmith35, 2024-05-05:
@lanewei120 @bambu123 did the dump file provide any information to help find the buffer overrun issue? The debug analysis said that the buffer overrun is a potential security risk.

lanewei120, 2024-05-05:
@tsmith35 we are sorry for the slow response these days due to the International Workers' Day holiday
we will check it soon

lanewei120, 2024-05-06:
@tsmith35 

thanks for uploading these files

could you also help to upload following dlls for our debugging?
C:\Windows\System32\ucrtbase.dll
C:\Windows\System32\ntdll.dll

furthermore
1. is there any previous version working with no crash?
2. if don't install the networking plugin (you can delete C:\user.name\Administrator\AppData\Roaming\BambuStudioBeta\plugins, and delete C:\user.name\Administrator\AppData\Roaming\BambuStudioBeta\BambuStudio.conf, then run bambu-studio again, at the end of start page, don't select netwoking plugin), is this issue still can be reproduced?