bambulab/BambuStudio #3957: macOS: Compilation of Boost fails

queengooborg, 2024-04-26:
### Bambu Studio Version

Local checkout of cfb273c84a229af86a3b926dd51baa779de93f9b

### Where is the application from?

Built from source by yourself or others

### OS version

macOS Sonoma 14.4.1

### Additional system information

MacBook Pro 16-inch 2021, Apple M1 Max, 64GB RAM

### Printer

Bambu X1 Carbon

### How to reproduce

When attempting to perform manual compilation of Bambu Studio to attempt to fix a bug, I had issues with the compilation process.  (I still am, but I'm slowly working through the issues.)

One issue happens to be with the frozen version of Boost.  Boost 1.78 uses [`std::piecewise_construct`](https://en.cppreference.com/w/cpp/utility/piecewise_construct) which has been replaced with `std::piecewise_construct_t` in C++17 -- and in turn, the macOS SDK.

### Actual results

```
In file included from ./boost/unordered/detail/implementation.hpp:44:
./boost/unordered/detail/fwd.hpp:53:16: error: no member named 'piecewise_construct' in namespace 'std'; did you mean 'piecewise_construct_t'?
    using std::piecewise_construct;
          ~~~~~^~~~~~~~~~~~~~~~~~~
               piecewise_construct_t
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__utility/piecewise_construct.h:20:29: note: 'piecewise_construct_t' declared here
struct _LIBCPP_TEMPLATE_VIS piecewise_construct_t { explicit piecewise_construct_t() = default; };
                            ^
```

### Expected results

Naturally, the project should have compiled without issue!

I've found that by upgrading to the latest version (at the time of writing) of Boost, Boost 1.85, it compiles just fine.  You can apply the following diff to `deps/Boost/Boost.cmake` to fix the issue:

```diff
ExternalProject_Add(
    dep_Boost
-   URL "https://github.com/bambulab/boost/releases/download/1.78.0/boost_1_78_0.zip"
-   URL_HASH SHA256=f22143b5528e081123c3c5ed437e92f648fe69748e95fa6e2bd41484e2986cc3
+   URL "https://boostorg.jfrog.io/artifactory/main/release/1.85.0/source/boost_1_85_0.zip"
+   URL_HASH SHA256=e712fe7eb1b9ec37ac25102525412fb4d74e638996443944025791f48f29408a
    DOWNLOAD_DIR ${DEP_DOWNLOAD_DIR}/Boost
    CONFIGURE_COMMAND "${_bootstrap_cmd}"
-   PATCH_COMMAND ${_patch_command}
    BUILD_COMMAND "${_build_cmd}"
    BUILD_IN_SOURCE    ON
    INSTALL_COMMAND "${_install_cmd}"
)
```

### Project file & Debug log uploads

Not applicable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-04-28:
thanks for doing this

are you using the default compile commands following guide
https://github.com/bambulab/BambuStudio/wiki/Mac-Compile-Guide?

did you change some settings?

queengooborg, 2024-04-28:
I have not changed any settings, no.  I tried to change some of the settings (such as setting the C++ version to 11) to see if that did the trick, but no luck there.

I believe that this is a side effect of using the SDK for a newer macOS version than the one Bambu Lab is compiled on for testing and distribution.  I have not researched this to confirm, but my theory is that the newer SDK has raised the minimum C++ version to 17 and dropped support for C++11 features altogether, and as a result, the currently-frozen version of Boost us unable to compile since it uses C++11 features.

I'm currently working on a PR to upgrade Boost and fix some of the other issues I've run into when trying to compile (they are probably all related to C++11 -> C++17).  I hope to have it finished and ready for review soon!

lanewei120, 2024-04-28:
so it should be related with the newer macOS

we will prepare an environment with new macos sdk and also tried here

thanks

queengooborg, 2024-04-28:
I've updated this issue to track more than just the Boost compilation failure.  I just submitted a draft PR, #3968, which includes the changes I have made so far!