bambulab/BambuStudio #3959: model is completely messed up when I run slice plate

mikevanoostende, 2024-04-26:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

mac os

### Printer

X1

### How to reproduce

upload stl file, doesn't matter which quality, then when i select "slice plate" the model is completely ruined 
I did not even have to print it, problem is with the software.

[image]
[image]
[image]
[image]

### Actual results

This used to work fine with other versions of bambulab...

### Expected results

proper slicing

### Project file & Debug log uploads

[tubes_18-12-23.STL.zip](https://github.com/bambulab/BambuStudio/files/15132469/tubes_18-12-23.STL.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

mikevanoostende, 2024-04-26:
Dear, 

I've done a restart of the software, and now it works again, not sure what happend the first time, so i guess.. you can close this .

lanewei120, 2024-04-28:
thanks for doing this

let's reopen it after it can be reproduced

closed 2024-04-28 by lanewei120