bambulab/BambuStudio #3960: Painful Object Filament Selection

chrisjenx, 2024-04-26:
**Is your feature request related to a problem? Please describe.**
Changing filament is awful ux

**Describe the solution you'd like**
The drop down filament selector shouldn't require 3/4 slow clicks to change. 1 to show drop down 1 to select the filament.

**Describe alternatives you've considered**
Right click is ok, but still cumbersome 

**Additional context**
Basically just make the selecting filament on the object pane much quicker to change than painful slow multi clicks.

OSx
Latest build

majianjia, 2024-04-27:
You can use number keys from the keyboard to set the filament, too. Did you try that?

chrisjenx, 2024-04-27:
Haven't tried.