bambulab/BambuStudio #3961: Typo in tooltip: Internal solid infill pattern

patridge, 2024-04-26:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

N/A

### Printer

Bambu Lab A1 mini

### How to reproduce

Hover the **Internal solid infill pattern** value field until the tooltip shows up.

The word "narrow" will be misspelled as "nattow". Possibly a result of one of these two lines:
* https://github.com/bambulab/BambuStudio/blob/cfb273c84a229af86a3b926dd51baa779de93f9b/src/libslic3r/PrintConfig.cpp#L1179
* https://github.com/bambulab/BambuStudio/blob/cfb273c84a229af86a3b926dd51baa779de93f9b/bbl/i18n/BambuStudio.pot#L7974

### Actual results

Tooltip text: "Line pattern of internal solid infill. if the detect nattow internal solid infill be enabled, the concentric pattern will be used for the small area."

### Expected results

Tooltip text: "Line pattern of internal solid infill. if the detect narrow internal solid infill be enabled, the concentric pattern will be used for the small area."

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

murongguao, 2024-05-06:
Thank you very much for your question, it will be fixed.

andypiper, 2024-05-07:
Confirmed here as well, came to report the same thing, noticed today.

[image]

DanBao-Bambu, 2024-05-15:
@andypiper Your requirements have been recorded.

closed 2024-05-15 by DanBao-Bambu