bambulab/BambuStudio #3963: Beta 4 - AMS Switch from 3rd party filament with PA to Bambu filament should auto reset.

TinyMito, 2024-04-26:
### Bambu Studio Version

1.9.1.58

### Where is the application from?

Bambu Lab github releases

### OS version

macOS

### Additional system information

Macbook Pro 14 M1 Pro

### Printer

P1S Combo

### How to reproduce

1. Add 3rd party filament.
[image]

2. Add custom PA profile to it.
[image]

3. Remove the 3rd party filament.
4. Put Bambu Filament brand with RF tag.

### Actual results

The issue here, the Bambu filament would still remain with the custom PA profile in which it should not be one. Should be default to blank (0.020) PA.

The actual model printed, is over extruded on corners and I found this out why.

[image]

[image]

### Expected results

After clicking Reset: back to normal. This should auto reset by default if RF tag filament is read.

[image]

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [X] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-04-28:
@TinyMito  Thanks for reporting this issue, we will fix it in the future.

gaolegao-lx, 2024-08-07:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

RyanEwen, 2024-10-09:
I run into this frequently and hope to see the PA Profile reset when a new filament is identified.