bambulab/BambuStudio #3964: Bambu Studio generates frequent hardware interrupts

AnalogKnight, 2024-04-27:
### Bambu Studio Version

1.9.0.60

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 23H2

### Additional system information

CPU: AMD Ryzen 9 3950X
RAM: 128GB
GPU: Nvidia RTX 3090

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. Just run Bambu Studio

### Actual results

When Bambu Studio is running, it triggers frequent hardware interrupts causing lag at the system level.

I'm sorry I can't provide steps to reproduce this, it happens 100% of the time on my computer. Whenever I run Bambu Studio, it continuously triggers hardware interrupts, which causes me to be unable to use any other application due to lagging, even if the CPU usage is below 10%.

Please let me know if any information is needed as I am not sure what kind of information to provide.

### Expected results

Hardware interrupts should not occur, at least not all the time.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/15137980/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-04-28:
@AnalogKnight 
could you tell us how to monitor a hardware interrupt?

and if from bambbustudio start
if you don't enter the device page,
does it happen?

AnalogKnight, 2024-04-28:
> @AnalogKnight could you tell us how to monitor a hardware interrupt?
> 
> and if from bambbustudio start if you don't enter the device page, does it happen?

Thank you for your reply.
I'm sorting the processes by hardware interrupts in reverse order in the Windows Resource Monitor and Bambu Studio is always at the top.
I noticed that Windows Memory Compression was also generating hardware interrupts when the problem occurred.
I'm experiencing the same problem in a virtual machine running on the same computer, so I don't think it's due to other software influences.
Maybe it's hardware related, but I'm not sure. My hardware specs aren't bad, and I haven't had problems with doing any other work.

If I'm on "preparing" page, the problem isn't so obvious, but it still happens. The problem is most noticeable in the "Preview" and "Devices" page.
It doesn't last all the time, so sometimes I can use Bambu Studio normally, but when it does, the hardware interruption makes it almost impossible to open Task Manager. When I close Bambu Studio, my system immediately shakes off the lag.

OzDruiD, 2024-04-28:
@AnalogKnight  Curiously I am tracking some recent lag in my systems as well but unable to isolate it.
Can you confirm its System Interrupts your looking at?  I have incluided a screenshot of my system running a long print, Bambu Studio open and no system interrupts showing....
[image]

lanewei120, 2024-04-28:
@AnalogKnight 
could you try this version https://github.com/bambulab/BambuStudio/releases/tag/v01.09.01.58?

we have improved a bit on this issue in recent version

AnalogKnight, 2024-04-28:
@OzDruiD 

I'll try to take a screenshot, the problem doesn't happen all the time when I use Bambu Studio, so maybe I'm mistaken.
What is certain though is that Bambu Studio does cause the system to lag, because when it is closed the system immediately returns to being responsive.

AnalogKnight, 2024-04-28:
@lanewei120 
Thanks, I'll try it and will update here later.

AnalogKnight, 2024-05-01:
@lanewei120 
The problem is the same, whenever I'm on the Preview or Devices page, my computer gets very laggy.
I've noticed that hardware interruptions don't happen 100% of the time, so maybe I'm mistaken. Sometimes the lag occurs with a hardware interrupt, other times it doesn't.
But I can't think of any other reason, my CPU usage is below 10% and only 40GB of my 128GB RAM is being used. What other reasons could cause this level of lag? Sometimes my system almost stops responding and opening the task manager is difficult. And when I end the Bambu Studio process in Task Manager, everything returns to normal.

lanewei120, 2024-05-06:
> @lanewei120 The problem is the same, whenever I'm on the Preview or Devices page, my computer gets very laggy. I've noticed that hardware interruptions don't happen 100% of the time, so maybe I'm [...]

we are sorry for the slow response these days due to the International Workers' Day holiday

do you mean there is no difference between 1.9.1.58?
and when using 1.9.1.58, when bambu-studio is in background, is the system still lagging?

nesttle, 2024-08-10:
Same problem for months, PageFault, same with Orca.

AnalogKnight, 2024-08-10:
I'm not sure if anyone is still following this Issue.

If I think correctly, Bambu Studio uses Edge WebView2, right?
I've found that the problem is especially noticeable when Microsoft Edge and Bambu Studio are running together. But I can't figure out if they are related.
The hardware interrupts seem to be generated by Microsoft Edge, but when I close Bambu Studio, Edge no longer generates hardware interrupts.

walterwongbbl, 2024-08-13:
@nesttle  @AnalogKnight 
We are currently fixing the issue with the PageFault.