bambulab/BambuStudio #3966: Calibration Screen Freezes

marcellatwin, 2024-04-27:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 12.7.4

### Additional system information

MacBook Pro (Retina, 13-inch, Early 2015)
Processor:  3.1 GHz Dual-Core Intel Core i7
Memory:  16 GB 1867 MHz DDR3
Graphics:  Intel Iris Graphics 6100 1536 MB

### Printer

Bambu Lab P1S

### How to reproduce

Run Calibration Flow Rate test.
Once test is done, issue occurs.
screen shows either complete or N/A and Next button is doesn't proceed forward.  This has happened on m

### Actual results

Screen shows either test complete or N/A and Next button is doesn't proceed forward.  This has happened on both Calibration 1 and 2.  Requiring me to rerun the test.  Happened 2 out of the 4 times I ran this test.

### Expected results

It should move to another screen after hitting next, where you can select which of the calibration tiles was the smoothest.

### Project file & Debug log uploads

[Testing.3mf.zip](https://github.com/bambulab/BambuStudio/files/15139536/Testing.3mf.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/15139537/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-04-28:
@marcellatwin 
1. What is your printer firmware version?
2. Dose "Next button is doesn't proceed forward" means the next button is grayed out and cannot be clicked, or there is no response when clicking the next button?

TMillesich, 2024-04-28:
I can reproduce it too in line and shape-mode, but it does not happen always. *

Firmware: P1 series OTA version 01.05.02.00
LAN Mode: Enabled
Bambu Studio: v01.08.04.51
Windows 10 22H2

The Button is greyed out/disabled.

* There's a high change for it when you forget that the calibration print finished and keep the dialog open for 5-10min.

marcellatwin, 2024-04-28:
Sorry for the late reply.  So to answer your questions:
-Printer Version:  01.05.02.00
-Greyed out and could not but clicked
-I believe yes you are right, in terms of a timeout sort of situation.
First time it happened to me, yes it took a while for me to remove the
calibration print and look at it closely.  And by the time I got back to
the computer to select, some time had passed.  HOWEVER, it was only 60-90
secs for me the next time around, but my theory is as follows:

-I was on the Device tab watching the print.  Once it had finished, I
walked (actually jogged) out to the garage, where my printer is, grabbed
the print, selected the best calibration and ran back.  At the computer I
then selected the Calibration tab, THEN I discovered the greyed out button.

-The next time, thinking that might be the issue, I left the window open on
the Calibration tab and watched the print on my phone Bambu App.  So my
theory is you can't leave the tab AS IT FINISHES.  Before it finishes you
can switch between the tabs but can't do it when the "calibration next
step" code or something is loading.

-I haven't been able to test this one more time so it is just a theory.

Jonathan Crossley