bambulab/BambuStudio #3970: Slicer Preview freezes after inserting PAUSE command

PhilR75, 2024-04-28:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10, 22H2, 64 Bit

### Additional system information

NVidia GeForce RTX 2080, Game Ready Driver 552.12

### Printer

Bambulab X1C

### How to reproduce

1. Load a model (does not matter if it is a simple or very complex model)
2. Go to "Preview" and slice build plate
3. Go to any layer, right click on the position button and "insert pause"

### Actual results

[image]

Screen freezes and moving the layer preview to insert another pause is no longer possible without re-slicing the buildplate

### Expected results

[image]

1) Moving the slider to insert another pause is still possible
2) re-slicing the build plate before printing is required

I know that re-slicing is required before printing, but as I manually want to change the filament multiple times, being able to insert multiple pauses without having to re-slice would be nice. 

### Project file & Debug log uploads

[project.zip](https://github.com/bambulab/BambuStudio/files/15141673/project.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/15141674/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

simar1998, 2024-04-30:
When you go to prepare and go back to the preview page the pause the layer slider can be used again.

LiZ-Li-BBL, 2024-06-12:
This is not a 100% reproducible issue. We are trying to reproduce this problem and will fix it in a future version.