bambulab/BambuStudio #3972: Dimensional inaccuracies with cylindrical/round objects

Schokobecher, 2024-04-28:
### Bambu Studio Version

1.9.1.58

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.4

### Printer

Bambu Lab X1C

### How to reproduce

1. Open new projects in BS
2. Add Primitive --> Cylinder
3. Leave Presets @ Default (Generic Filament, 0.2mm Standard Profile)
4. Print
5. Measure the cylinder with calipers

### Actual results

Cylindrical/round objects typically come out -0.2 mm in X/Y from the dimensions in BS. Other features/parts are not affected by this.

This has been something I've been observing for a while now, so it's not new to the 1.9 release.
TPU, PP, PLA, and PETG all behave similar in my tests.

More people seem to be affected by this:
https://www.reddit.com/r/BambuLab/comments/15kw4oc/bambulab_p1p_dimensional_accuracy_issue/

The constant factor of -0.2 mm makes me believe that this is a slicing issue, as it does not scale with the model, or material used.

### Expected results

Parts should match the dimesions in the slicer as much as the FDM prcoess allows.

### Project file & Debug log uploads

[Cylinder.3mf.zip](https://github.com/bambulab/BambuStudio/files/15141920/Cylinder.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-04-30:
This appears to be related to filament shrinkage. Different shapes will affect the shrinkage effect. We will discuss it and try to solve it.

insane78, 2024-05-01:
> This appears to be related to filament shrinkage. Different shapes will affect the shrinkage effect. We will discuss it and try to solve it.
It's not shrinkage, i tried with many kind of pla and same result... tried with a voron with prusaslicer and result perfect with same pla

Schokobecher, 2024-05-01:
I have to disagree @QingZhangBambu 
While shrinkage definitely can have an impact, a 0.2 mm flat deviation is not possible. 

I first noticed it with a PP print, which can have significantly shrinkage (1-3%), but the results do not scale with the object size. My TPU 95A (shrinkage of 0.8-1.8% typically) print have the same 0.2 mm deviation as my PP prints, but both models are vastly different geometry and sizes.

Shrinkage would be a percentage of the model diameter, not a flat value, and also would affect other models (Cube vs. Cylinder)
If it helps, I can take more measurements with different materials/brands/colors and model sizes, and report back.

QingZhangBambu, 2024-05-06:
i've checked slicer date, it data is fine. for example, the circle is 25mm, there is the data
[image]

QingZhangBambu, 2024-05-06:
I printed a cylinder with a diameter of 25.6 mm and the result was 25.44 mm.
I printed two cylinders simultaneously, one using a wall speed of 20mm/sec and a infill speed of 50mm/sec, the diameter of the slower one is almost 25.58mm. The other is 25.48mm
[圆柱收缩0.2的测试.zip](https://github.com/bambulab/BambuStudio/files/15215562/0.2.zip)

Schokobecher, 2024-05-06:
Is there anything that can be done besides limiting wall print speed to 1/10th of the default?
Default preset has wall speeds of 200/300 and Infill of 270

The 25.58 is definitely acceptable for PLA

QingZhangBambu, 2024-05-07:
@Schokobecher I just set the speed parameters. Concerned staff recommend cleaning the carbon rod first and then coating it with liquid paraffin, which can help you get better results at the default speed. You can try these first.
[image]

Schokobecher, 2024-05-07:
@QingZhangBambu I'll give the rods a deep clean and apply some liquid paraffin as suggested and report back.

Just to confirm, the paraffin will not mess with the self-lubricating bearings?

> Bambu Lab printers utilize self-lubricating bearings. However, after the cleanup process is complete, it may take some time for the lubrication to reach the desired level.
> 
> To ensure proper lubrication and calibration of the printer, perform an additional 10-20 movements of the print head along the entire length of the dry carbon fiber rod. This step guarantees that the lubrication functions as intended and allows the printer to adjust itself according to the new resistance encountered.
> 
> Usually, this step is not necessary for every operation, but it can be beneficial in certain situations.

https://wiki.bambulab.com/en/general/carbon-rods-clearance

QingZhangBambu, 2024-05-09:
@Schokobecher I will confirm paraffin things with the concerned staff. And I talked to the concerned staff about that issue, they are trying to solve it.

fly2016-1-12, 2024-05-09:
@Schokobecher  Our solution is being optimized and tested，but it cannot be provided in the short term, pls pay attention to future firmware updates~

wineBambu, 2024-05-09:
@Schokobecher Our solution is being optimized and tested，but it cannot be provided in the short term, pls pay attention to future firmware updates~

insane78, 2024-06-02:
This issue is really annoying, please consider to correct it as soon as possible. I printed a 30x30x30 cube with both my x1c and obtained x:29.84 and y:29.94. This is not a normal 3d printer tolerance. A fiend of mine with p1s doesn't have this problem...

Schokobecher, 2024-06-02:
@insane78 what your are seeing most likely does not relate to this issue, but rather is a calibration issue (Flow Rate, K Factor, Belts).
This issue so far only seems to affect round extrusions (holes have a general shrinkage)

gaolegao-lx, 2024-06-03:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

jordangarside, 2024-07-04:
I believe this issue in the forum is also related to this problem:
https://forum.bambulab.com/t/dimensionally-inaccurate-parts-being-produced-by-x1c/47984

hicwic, 2024-09-01:
Same problem as stated by OP.

Rectangular shapes are accurate on XY. Rounded shapes or circle a 0.2mm to small/tight on XY.

Any update on this ? or a workaround ?

hicwic, 2024-09-02:
For reference i did a 20mm cylinder stacked on a 20mm cube. Cube is accurate, cylinder not. (0.2mm deviation)

[image]
[image]
[image]

For testing:
- i tryed Orca, same result. So must be firmware related
- I setted shrinkage parameter to 99% (20mm => 19.8 => 1% deviation). Result : Cylinder is 20mm, cube is 20.20mm. So it's not a workaround.

hicwic, 2024-11-26:
any update on this ?
there is a lack in communication on bambu side...

hicwic, 2024-11-26:
> @Schokobecher Our solution is being optimized and tested，but it cannot be provided in the short term, pls pay attention to future firmware updates~

don't do this kind of anwser when you previously spoke about a solution being in testing state

MacieGx, 2024-12-24:
@QingZhangBambu looks like im having same issue. Do we have any updates on this?

QingZhangBambu, 2024-12-24:
@MacieGx The concerned staff are working hard to improve it.

MacieGx, 2024-12-24:
> @MacieGx The concerned staff are working hard to improve it.

Thank you for your prompt response. Could you please let me know if we can expect it sometime soon, for instance in Q1 2025, or if there is currently no estimated timeframe?

jordangarside, 2024-12-24:
For what it’s worth:
I basically recommend everyone I know against getting a Bambu printer until the issue is fixed.

An M10 bolt has around ~0.2mm of tolerance.
With cylinders coming out ~0.2mm too small, you basically can’t even size your model anywhere in the range of an M10 bolt’s tolerance (you need to do something like 10mm instead of 9.8mm, and the range is 9.7-9.9mm).

This means that if I want to share models that require dimensional accuracy with other people, then I need have to have specific models for bambu printers and so I end up just not sharing them.

It also leads me to not want to print anything like Gridfinity as I don’t know what type of round parts exactly trigger the issue and how much (i.e. does it happen on round corners?). If I ever get a printer that works correctly in the future, then the parts printed with the Bambu printers won’t fit nicely with the accurate printer.

0.2mm is about an order of magnitude more error than we would expect for something like a 10mm part, and the issue is worse when considering something like a 5mm part.

Also the longer this problem goes on the more a solution requires an option for the users to enable/disable the fix because otherwise their parts won’t fit nicely together with parts printed before the fix.

QingZhangBambu, 2024-12-26:
@MacieGx 
Sorry, I asked, but the exact time cannot be determined.
Quote from the concerned staff: "After testing and troubleshooting, we found that the problem was caused by motion lag caused by structural resistance. We will provide relevant calibration functions in subsequent firmware versions."

natec007, 2025-01-02:
Same issue, I have been printing Legos and the bumps are supposed to be 4.8mm circles, I add 0.2mm to them so they are 5mm in stl and they print 4.8mm every time.  Same results with multiple brands of PLA including Bambu Labs PLA.  Always exactly 0.2mm smaller than drawing.

StirlingEngine, 2025-02-01:
P1S, same issue. Inaccuracy on round objects beyond any shrinkage, almost stable -0.2mm on any size

shynie, 2025-02-01:
Same with P1S. Is there any news on the firmware update?

aabd2, 2025-04-03:
It's been almost 1 year and Bambulabs have not done anything?

Tread-On-Them, 2025-05-14:
I am encountering the exact same issue with a 18mm dia. cylinder. Interesting to note that conical shapes (those with overhangs etc.) are not affected for me, it only occurs with vertical cylindrical geometry.

soxhi8, 2025-11-18:
I have been consulting many people on which printer to get and I always clearly state that Bambu Lab printers are great for non-functional decorations. Functional parts with accurate dimensions are something that an Ender 3 can do, but apparently Bambu Lab cannot.

I am looking forward to a fix. Most people have a hard time believing that a brand such as Bambu Lab would produce such issues, rightfully so.

DragonB-bambu, 2026-09-20:
I'd like to clarify some maintenance points regarding this issue:Please do not apply any grease, oil, or wax to the carbon fiber rods. Doing so can cause permanent damage to the motion system. Please keep the rods clean and dry. Instead, we recommend checking and adjusting your belt tension as part of your regular maintenance. Ifyour machine has been in use for a long time, we also recommend replacing the belts to ensure optimal precision.
If your model coordinates are correct in the slicer but the print is uIndersized, please consider that this is often caused by factors such as belt elongation over time and the natural shrinkage of the filament material.

ComputerRing, 2026-09-20:
This is a useless reply, sorry. The issue has already been acknowledged by your team as FW issue, and a consistent 0.2mm deviation regardless of filament type or brand (and thus shrinkage) was already proven.