bambulab/BambuStudio #3973: Ubuntu 24.04 LTS: error while loading shared libraries: libwebkit2gtk-4.0.so.37

mhoellein, 2024-04-28:
### Bambu Studio Version

1.9.1 Public Beta 4

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04

### Printer

Bambu Lab X1

### How to reproduce

start the AppImage

### Actual results

error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Expected results

BambuStudio should start

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/15143098/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Bjoernsen, 2024-04-29:
I recently updated from 23.10 to 24.04 and have exactly the same error message. Everything worked with 23.10. I already  install `libwebkit2gtk-4.1-dev` (4.0 does not exist) via apt without success.

mampaiva, 2024-04-30:
I solved this problem by adding the previous lts repository and installing version 4.0. 
For this I changed the file sudo vim /etc/apt/sources.list.d/ubuntu.sources
Add the follow line: 
`Types: deb
URIs: http://br.archive.ubuntu.com/ubuntu/
Suites: jammy noble-updates noble-backports
Components: main restricted universe multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg

Types: deb
URIs: http://security.ubuntu.com/ubuntu/
Suites: jammy-security
Components: main restricted universe multiverse
Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg`

then update the repositories with apt update and go to synaptic to search and install the correct version of libwebkit2gtk

[image]

ajft, 2024-04-30:
thanks @mampaiva - you've solved a problem I'm experiencing with Cisco AnyConnect VPN software that gives the same webkit error

sanity, 2024-05-06:
Recently updated to Ubuntu 24.04 and experiencing the same issue.

lanewei120, 2024-05-07:
please try following commands on ubuntu 24.04 to solve this issue currently

```
sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
sudo ln -sf /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18
```

we are also trying to solve this issue without the above workarounds

LuisPortanel, 2024-05-10:
Thanks, @mampaiva . I recently upgraded to Ubuntu 24.04 and I had issues with my Pulse Secure VPN / Ivanti Secure VPN installation. With your workaround, I could finally install the VPN. 
Thanks!

mechano, 2024-05-11:
Creating the links doesn't work on my Ubuntu 24.04 

```
sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
sudo ln -sf /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18

```
I've a core dump.

But it works for me this command from terminal:

`WEBKIT_DISABLE_COMPOSITING_MODE=1 .appImage/OrcaSlicer.AppImage`

I've added to the desktop icon I've created under .local/share/applications

```
[Desktop Entry]
Name=OrcaSlicer
GenericName=3D Printing Software
Icon=OrcaSlicer
Exec="WEBKIT_DISABLE_COMPOSITING_MODE=1 /home/mechano/.appImage/OrcaSlicer.AppImage" %F
Terminal=false
Type=Application
MimeType=model/stl;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;application/x-amf;
Categories=Graphics;3DGraphics;Engineering;
Keywords=3D;Printing;Slicer;slice;3D;printer;convert;gcode;stl;obj;amf;SLA
StartupNotify=false
StartupWMClass=OrcaSlicer
```

lanewei120, 2024-05-13:
I have also added those webkit libraries into appImage which can be executed without any modification
here is the link
https://github.com/bambulab/BambuStudio/actions/runs/9040304943/artifacts/1493056554

DipFlip, 2024-05-14:
> please try following commands on ubuntu 24.04 to solve this issue currently
>
> ```
> sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
> sudo [...]

This results in the following error for me (Ubuntu 24.04)

`(process:79503): libsoup-ERROR **: 13:25:25.773: libsoup2 symbols detected. Using libsoup2 and libsoup3 in the same process is not supported.`

lanewei120, 2024-05-15:
> I have also added those webkit libraries into appImage which can be executed without any modification here is the link https://github.com/bambulab/BambuStudio/actions/runs/9040304943/artifacts/1493056554

@DipFlip how about trying this appImage which includes the libwebkit libraries

DipFlip, 2024-05-15:
> > I have also added those webkit libraries into appImage which can be executed without any modification here is the link https://github.com/bambulab/BambuStudio/actions/runs/9040304943/artifacts/1493056554
> 
> @DipFlip how about trying this appImage which includes the libwebkit libraries

Thank you @lanewei120 but I'm actually trying to run another program - Cisco Anyconnect. It also throws the same error so that's how I ended up on this thread.

jinzzasol, 2024-05-24:
> I solved this problem by adding the previous lts repository and installing version 4.0. For this I changed the file sudo vim /etc/apt/sources.list.d/ubuntu.sources Add the follow line: `Types: deb URIs: http://br.archive.ubuntu.com/ubuntu/ Suites: jammy noble-updates noble-backports Components: main restricted universe multiverse Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg
> 
> Types: deb URIs: http://security.ubuntu.com/ubuntu/ Suites: jammy-security Components: main restricted universe multiverse Signed-By: /usr/share/keyrings/ubuntu-archive-keyring.gpg`
> 
> then update the repositories with apt update and go to synaptic to search and install the correct version of libwebkit2gtk
> 
> ![Screenshot from 2024-04-30 11-27-03](https://private-user-images.githubusercontent.com/64433509/326847788-ff1a0371-5f56-4805-a818-d77cd3f0f474.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY1NjIzODUsIm5iZiI6MTcxNjU2MjA4NSwicGF0aCI6Ii82NDQzMzUwOS8zMjY4NDc3ODgtZmYxYTAzNzEtNWY1Ni00ODA1LWE4MTgtZDc3Y2QzZjBmNDc0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI0VDE0NDgwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI5ZjQxY2I0YjIyMGYzYmNjMzgwNmU0ZjI0YjM4YzMyOGM1MTkxM2ZhNDUxNDZiZmIxZDNmMDJjYWExNzU3MTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Q1Pe_aOdr2Iug8KmtyTDUYwphbKcLD5fysanO4VvDRE)

This is the best solution. Thank you.

MackBambu, 2024-05-30:
On Ubuntu, each release has its own versions of software. When a new LTS release is published, older releases no longer receive updates for their software. However, our software cannot operate in this manner. Supporting an increasing number of Linux distributions can lead to larger and more cumbersome software packages.
I propose creating a script that detects different distributions and corrects their inconsistencies.

thaowl, 2024-06-08:
Using  building on Ubuntu 24.04 https://github.com/tauri-apps/tauri/issues/9662 helps. 

--
I worked around this by adding the following line to /etc/apt/sources.list:

deb http://gb.archive.ubuntu.com/ubuntu jammy main

Then doing:

sudo apt update
sudo apt install libwebkit2gtk-4.0-dev

--
remove added line from sources.list after installation to keep enviroment sane.

Then build deps and BambuStudio. 
Altough deps do build with libwebkit2gtk-4.1, building BambuStudio later will fail.
Works for me.

MWWorks, 2024-06-13:
Confirm that the new 1.9.2 release also has this issue, it is unable to run on eg Ubuntu 24.04.

Also confirm that the patched version provided above by @lanewei120 seems to work - so please patch the 1.9.2 - and ideally include it going forward, the appimage should have all dependencies :)

caco3, 2024-06-14:
Same issue for me on Ubuntu 24.04

hainjedaf, 2024-06-16:
Same issue on Debian unstable/Sid/Trixie

since 1.09 no BambuSLicer runs. All crash with following error:
[code]
/tmp/.mount_bambu-L82HJc/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
[/code]

Unpacking the AppImage and running the executable, causes same error

lanewei120, 2024-06-17:
> Confirm that the new 1.9.2 release also has this issue, it is unable to run on eg Ubuntu 24.04.
> 
> Also confirm that the patched version provided above by @lanewei120 seems to work - so please patch the 1.9.2 - and ideally include it going forward, the appimage should have all dependencies :)

that version can not work on some linux versions also, we will find a final solution later

currently please use the workaround solution or the flathub version https://flathub.org/apps/com.bambulab.BambuStudio

Andrew-Pynch, 2024-06-22:
@mampaiva this worked, thank you so much!

MackBambu, 2024-06-27:
https://github.com/MackBambu/BambuStudio/actions/runs/9691686545/artifacts/1643561459
This appimage was built using libwebkit2gtk-4.1.

leodrostes, 2024-07-26:
> I solved this problem by adding the previous lts repository and installing version 4.0. For this I changed the file sudo vim /etc/apt/sources.list.d/ubuntu.sources Add the follow line: `Types: deb [...]

it worked for me too. Thanks!

m1ndcr4ck, 2024-08-14:
> Thank you @lanewei120 but I'm actually trying to run another program - Cisco Anyconnect. It also throws the same error so that's how I ended up on this thread.

@DipFlip , I'm trying the same, any luck to solve this issue?

DipFlip, 2024-08-14:
> @DipFlip , I'm trying the same, any luck to solve this issue?

@m1ndcr4ck yes [thaowl](https://github.com/thaowl)'s solution worked for me so now i have Anyconnect installed on Ubuntu 24.04 and it's working fine!

m1ndcr4ck, 2024-08-15:
> @m1ndcr4ck yes [thaowl](https://github.com/thaowl)'s solution worked for me so now i have Anyconnect installed on Ubuntu 24.04 and it's working fine!

@DipFlip  thanks so much, currently, I was able to run it using this workaround https://github.com/bambulab/BambuStudio/issues/3973#issuecomment-2085476683, however, I still need it on a different Linux distro "Kali"

jmcbri, 2024-09-15:
> Thank you @lanewei120 but I'm actually trying to run another program - Cisco Anyconnect. It also throws the same error so that's how I ended up on this thread.

DipFlip - Did you get Citrix to run?  If so, how. (Sorry!)

xandark, 2024-10-13:
My fix was simple and didn't need to mess with changing repository sources, rather in 24.04 I just installed one package:

`sudo apt install libwebkit2gtk-4.1-0`

CrustyBarnacle, 2025-01-03:
For any, like myself, still wrangling this... the flathub version works:
- https://flathub.org/apps/com.bambulab.BambuStudio

kazelin, 2025-08-21:
> My fix was simple and didn't need to mess with changing repository sources, rather in 24.04 I just installed one package:
> 
> `sudo apt install libwebkit2gtk-4.1-0`

This solved it for me on Ach 6.16.0-arch2-1

`yay -S webkit2gtk-4.1 2.48.5-1`

kalinjul, 2025-09-27:
This issue is appearing when running ubuntu 25.04 trying to use  Bambu_Studio_ubuntu-24.04_PR-8184.AppImage.
Shouldn't the app image include all required libraries?