bambulab/BambuStudio #3981: Filament changing still happens after canceling part on the build plate

kyeager95, 2024-04-30:
### Bambu Studio Version

1.8.4.5.1

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

Have 2 models being printed both using support material that is not the same as the model material. the 1 model is has features that require support material are taller than the other model. You cancel the taller model due to finding a failure due to warping.

### Actual results

The machine will stop working on the taller model, but it will keep filament changing when it is no longer required since you are past the point of need support material on the shorter model.

### Expected results

Stop wasting filament thru the poop chute by stop changing the filament and doing the prime tower since it is no longer needed.

### Project file & Debug log uploads

[support issue files.zip](https://github.com/bambulab/BambuStudio/files/15158572/support.issue.files.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2024-05-04:
do you mean skip the object from bambu handy or printer's screen?

kyeager95, 2024-05-04:
Skip the object on the printer screen.

SaltWei, 2024-05-06:
[image]
Not a bug

kyeager95, 2024-05-06:
Ahh okay, I did not know that thank you.

SaltWei, 2024-08-06:
After further investigating, it's impossible to skip filament changing while keep color no-mixing when printing. So won't do at the moment

closed 2024-08-06 by SaltWei