bambulab/BambuStudio #3982: Setting Top Z Distance with Tree Hybrid creates missing layers

AJZein, 2024-04-30:
### Bambu Studio Version

01.09.01.58

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

Bambu Lab P1P

### How to reproduce

Slice the included .3mf file. Look around layers 78 and 294 to see the mentioned missing layers. Reset Top Z distance or uncheck Independent Support Layer Height and slicing returns to normal.

### Actual results

The missing layers caused the model to split at those points. Below are images of the missing layers in the sliced model and the (full) printed model
[image]
[image]
[image]

### Expected results

No missing layers created when slicing

### Project file & Debug log uploads

[tree_support_lines_test.zip](https://github.com/bambulab/BambuStudio/files/15163101/tree_support_lines_test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2024-05-24:
Thanks for reporting the issue. We'll fix it soon.

AJZein, 2024-06-18:
Confirmed fixed in version 1.9.2.57

closed 2024-06-18 by AJZein