bambulab/BambuStudio #3985: [Issue] Libraries need to be updated to work with latest Ubuntu 24.04 LTS

BambuLabDiscord, 2024-05-01:
Fails to run on the current stable LTS release due to various missing / updated libraries. they need to be packaged with the appimage or the software updated to use newer versions.

lanewei120, 2024-05-04:
could you give the error log with all the missed libraries?

mzanetti, 2024-05-11:
/tmp/.mount_Bambu_g991M7/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

lanewei120, 2024-05-13:
the same issue as https://github.com/bambulab/BambuStudio/issues/3973
and we also gave solution in https://github.com/bambulab/BambuStudio/issues/3973

mzanetti, 2024-05-13:
Yeah, symlinking the newer libraries to the older seems to work as a temporary workaround, although it seems to cause Bambu Studio to crash occasionally...

ProxiBlue, 2024-07-29:
@lanewei120 

> and we also gave solution in https://github.com/bambulab/BambuStudio/issues/3973

it is more a workaround, than a solution ;) and it is far from a working.  
Someone did the effort to recompile this with the correct libs, but, not great to just run any unofficial appimage

lanewei120, 2024-07-29:
we have relased an appimage for ubuntu 24.04 specially
https://github.com/bambulab/BambuStudio/releases/download/v01.09.03.50/Bambu_Studio_linux_ubuntu_24.04_v01.09.03.50.AppImage

could you have a try
@ProxiBlue

ProxiBlue, 2024-07-29:
@lanewei120 : works :) Thank you very much.

Actions taken to ensure tested correctly:

* Removed the link workaround 

```sudo rm /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
sudo rm /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18```

* test current appimage - fails as expected with the lib error (confirmed back at broken image)

* Run new supplied image

Worked as expected.

Thank you